FreeCAD/FreeCAD #6698: [Bug]Graphics Issues in Some Linux Configurations

WandererFan, 2022-04-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://www.forum.freecadweb.org/viewtopic.php?f=35&t=66406

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 21.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: cb9dda708494f3d20ec153c7881b9795a550c265
Python 3.9.7, Qt 5.15.2, Coin 4.0.0, OCC 7.5.2
Locale: English/Canada (en_CA)
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

Missing vertex, skinny line and extra line problems have been reported in
Fedora KDE/Plasma Qt 5.15.2
Manjaro KDE/Plasma Qt 5.15.2
Arch Gnome Qt 5.15.2
Ubuntu Gnome Qt 5.15.2
Ubuntu KDE/plasma Qt 5.15.2

On the other hand, we have a Debian installation using Qt 5.15.2 that works fine. 

TechDraw is essentially unusable in this state. 

### Anything else?

[Additional forum discussion](https://www.forum.freecadweb.org/viewtopic.php?f=35&t=67267)

My initial thought was that this was a Wayland vs X problem, but running FC in a Ubuntu 20 VM generates graphics issues using either wayland or X. 

Next thought was Qt 5.15.2, but we have at least 1 user that is having success with this version of Qt. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-04-01:
further information...
On a Ubuntu 20/Gnome/Qt 5.15.2 VM, FreeCAD v019 from the PPA exhibits the same graphics errors.

luzpaz, 2022-04-03:
This issue is not present on latest Snap edge/nightly package. @WandererFan suspects it has to do with Qt5.15.2 since Snap is using Qt5.15.3

```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28609 (Git)
Build type: Release
Branch: master
Hash: d666cc411403b623b2caf0010207e706bea3d76b
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
Installed mods: 
  * ToolbarStyle
  * Assembly4 0.11.10
  * stickframe
  * sheetmetal 0.2.49
  * pinger
  * Assembly3 0.11.1
  * Behave-Dark-Colors 0.0.1
  * Curves 0.3.0
  * Ship
  * BIM 2021.12.0
  * A2plus 0.4.55a
```

WandererFan, 2022-04-03:
Qt 5.15.2 works in at least 1 case.  Waiting for local build of release version

From **stable PPA** - does not work
OS: Ubuntu 21.10 (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: 0d9536ed3e8c7f40197b5606e1b7873625e1d6fe
Python version: 3.9.6
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/Canada (en_CA)

From **daily PPA** - does not work
[code]
OS: Ubuntu 21.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: cb9dda708494f3d20ec153c7881b9795a550c265
Python 3.9.7, Qt 5.15.2, Coin 4.0.0, OCC 7.5.2
Locale: English/Canada (en_CA)
[/code]

**Local build** works, but gets message:
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[code]
OS: Ubuntu 21.10 (ubuntu:GNOME/ubuntu) (virtualbox VM)
Word size of FreeCAD: 64-bit
Version: 0.20.28608 (Git)
Build type: debug
Branch: master
Hash: 66d0c35dc780459be0253a990abe2c2afa8fa814
Python 3.9.7, Qt 5.15.2, Coin 4.0.0, OCC 7.5.2
Locale: English/Canada (en_CA)
[/code]

mfraser, 2022-04-03:
With the latest weekly AppImage build:
[code]
OS: Ubuntu 21.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28611 (Git) AppImage
Build type: Release
Branch: (HEAD detached at a3b804e)
Hash: a3b804e7806c2b3b5e1228c41bca4569a743b400
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.2.49
  * lattice2 1.0.0
  * fcgear 1.0.0
  * Curves 0.3.0
  * Assembly4 0.11.8
  * fasteners
  * A2plus 0.4.54b
  * VendorParts
[/code]

PDF export works correctly and changing the line thickness in the lines section of View actually changes the line thickness of the drawing.

WandererFan, 2022-04-03:
The Qt 5.15.2 theory is discredited.
Local release build works
[code]
OS: Ubuntu 21.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.28608 (Git)
Build type: Release
Branch: master
Hash: 66d0c35dc780459be0253a990abe2c2afa8fa814
Python 3.9.7, Qt 5.15.2, Coin 4.0.0, OCC 7.5.2
Locale: English/Canada (en_CA)
[/code]

mfraser, 2022-04-03:
So that would point to something in the PPA builds causing this?

WandererFan, 2022-04-03:
> So that would point to something in the PPA builds causing this?

I guess so, but I have no idea what it might be at this point.

luzpaz, 2022-04-03:
So what do we have left? 
python, OCC, coin ?

mfraser, 2022-04-03:
> So that would point to something in the PPA builds causing this? I will try a local build myself later.

Compiled it and the lines are visible.

[code]
OS: Ubuntu 21.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28613 (Git)
Build type: Unknown
Branch: master
Hash: ab2461629cb06e08441943709ece6eff3a662434
Python 3.9.7, Qt 5.15.2, Coin 4.0.0, OCC 7.5.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.2.49
  * lattice2 1.0.0
  * fcgear 1.0.0
  * Curves 0.3.0
  * Assembly4 0.11.8
  * fasteners
  * A2plus 0.4.54b
  * VendorParts
[/code]

Not sure what to suggest, Coin and OCC versions are the same. Different Python versions, but they seem unrelated to whether it works or not.

WandererFan, 2022-04-07:
[Help request posted to Packaging forum](https://forum.freecadweb.org/viewtopic.php?f=42&t=67842)

0penBrain, 2022-04-12:
Don't know at which point it can lead to such effect, but comparing library dependencies, I found that my self-compiled version (that works correctly) still depends on `libQt5OpenGL` while PPA version (that shows problem) doesn't -- but my PPA version is pretty old as I'm running Kubuntu 21.04 --.

mfraser, 2022-04-12:
https://wiki.freecadweb.org/Compile_on_Linux quotes libqt5opengl5-dev as a dependency for compiling.

Can't see libqt5opengl as a dependency for freecad-daily-python3 from the PPA:
freecad-daily-common, python3-matplotlib, python3-pivy, python3-ply, python3-pyqt5, python3-pyside2.qtcore, python3-pyside2.qtgui, python3-pyside2.qtuitools, python3-pyside2.qtsvg, python3-pyside2.qtwidgets, python3-pyside2.qtxml, libfreecad-daily-python3-0.19, libfreecad-daily-python3-0.19, python3-yaml, libc6, libgcc-s1, libpython3.9, libqt5core5a, libqt5gui5 | libqt5gui5-gles, libqt5widgets5, libstdc++6

WandererFan, 2022-04-14:
When started from terminal, local build issues "Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway."

The PPA build does not issue this message when started from terminal. 

Is this a clue?

0penBrain, 2022-04-14:
@WandererFan I have no such message (neither on self-compiled nor PPA) but still can reproduce the issue.
So I'd tend to say it is not related to issue.

donovaly, 2022-05-15:
What's the status? It is declared as blocker but as far as I see FreeCAD cannot do anything since it depends on a certain combination of third-party libraries. Right?

adrianinsaval, 2022-05-16:
so libqt5opengl wasn't the problem? Could it be graphics driver related?

0penBrain, 2022-05-16:
> Could it be graphics driver related?

Don't think so because on any platform where it fails, a local build succeeds ;)

donovaly, 2022-05-17:
so then it is no blocker

luzpaz, 2022-05-17:
This should be added to a 'Known Issues' list for downloaders to see when then install v0.20

adrianinsaval, 2022-05-17:
@PLChris

0penBrain, 2022-05-18:
> so then it is no blocker

IMO, asking a user to self-compile to get something working isn't really a good option :)

WandererFan, 2022-05-20:
I hate to say this, but this has me beat.  I can build the PPA packages locally using the Launchpad recipe, but from there I hit the infamous brick wall.  Is there somebody in the community that knows this stuff that could help out?

donovaly, 2022-05-20:
> asking a user to self-compile to get something working isn't really a good option :)

Please explain what is happening. When you compile FC on your own it works. So FC is not to blame but a certain combination of packages.
It would be good to know what combination of packages causes the issue otherwise I have no idea how to fix the issue.

WandererFan, 2022-05-20:
> If FreeCAD is installed using the packages of 0.20 available on the PPA
> there are issues with line width when exporting from TechDraw. If it is
> compiled locally, it works from exactly the same source.

Judging by what flashes past on the screen, the compile/link that debuild
performs to build the packages is not the same as the traditional
cmake+make compile/link.  I don't know if it uses different options, or
pulls in different libraries, or if this is just revealing a weakness in
the code.  Debuild makes 3 packages whereas the local compile makes 1 big
executable.

donovaly, 2022-05-21:
> If FreeCAD is installed using the packages of 0.20 available on the PPA there are issues with line width when exporting from TechDraw.

Can you please explain I understand it too? ;-)
What is PPA?: https://en.wikipedia.org/wiki/PPA
I guess you refer to this?: https://en.wikipedia.org/wiki/Personal_Package_Archive

If so, this is a distribution-specific thing. Can you please inform the persons who are responsible for the packages there?

In general, we can't hold the release for a distribution-specific issue because otherwise a release will never happen ;-).

adrianinsaval, 2022-05-21:
There is an official freecad PPA for various ubuntu versions mainly maintained by @kkremitzki I believe, but the issue is not limited to these because it was supposedly reproduced in 3 different independent families of distros (fedora, ubuntu,  arch/manjaro). I can't properly judge if this can be considered a release blocker but IMO it isn't, if we find a solution it might be done at the packaging level only without requiring a new release and if it requires changes in FreeCAD code we can just cut a new point release but the fact it works ok on local compilations suggests it's not a fault in FreeCAD's code.

donovaly, 2022-05-22:
Just to get it right:
sicne it works when users compile FC themselves, it always works -> so we have no bug in FC
Since it does not work for the Ubuntu PPA -> the bug must be in the packaging.
@kkremitzki can you please have a look?

ask6483, 2022-07-24:
I got this issue also on Debian bookworm/sid in version 0.19 installed from repository but also on 0.20 installed from sid repo - can't change any line width in TechDraw...

luzpaz, 2022-07-25:
Installing FreeCAD from Debian or Ubuntu repos is not a good idea. Packaging in these repos is quite difficult and not uptodate. Better to utilize [Snap](https://wiki.freecad.org/Snap) or [Flatpak](https://wiki.freecad.org/flatpak) now.

rabfulton, 2022-08-13:
I had this issue on Manjaro/Arch linux with the 0.19 version. Since the updated packages have been released (v0.20) it is working now for me. If it helps I think some patches were removed from the build by the maintainer.

norru, 2022-08-28:
Just had this problem on Devuan Ceres.

I hope I don't have to switch to the FlatPak version because the font rendering on that one is all over the place :(

phrxmd, 2022-08-29:
I still have this issue with 0.20.1 on OpenSUSE Tumbleweed, which is a RPM-based distro that does not use DEB or PPAs.
I would use the AppImage instead, but that doesn't support Wayland.

The distribution's package for FreeCAD with the SPEC file used for building is at https://build.opensuse.org/package/show/openSUSE%3AFactory/FreeCAD

adrianinsaval, 2022-08-29:
does the appimage fail in wayland? AFAIK freecad itself doesn't have support for wayland and always runs under xwayland

phrxmd, 2022-08-29:
> does the appimage fail in wayland? 

The AppImage version always runs under XWayland.

The distribution version will run under Wayland, but I can force it to run under XWayland by specifying `QT_QPA_PLATFORM=xcb` on the command line. However, the issue arises whether I start it under Wayland or not.

> AFAIK freecad itself doesn't have support for wayland and always runs under xwayland

I think the issue here is with the AppImage and whether it has the QT Wayland platform plugin compiled in or not. If it has only the XCB platform plugin, then it will only support XWayland. This is a common issue with AppImages.

norru, 2022-08-29:
I have used both the distro package and FlatPak. The FlatPak version has huge font rendering issues. In a significant amount of casse, text can be eiither too small to be unreadable,or so large that only a small amount of text can fit in the screen.

The distro package has no font rendering issues but I get those other odd bugs, such as lines not being printed.

I basically have to have both versions and depending on whichever one has a showstopper bug, switch to the other hoping that the project files get parsed correctly (In older version, FlatPak saves wouldn't be read by the package and viceversa).
 
I'm using version 0.20 on Xfce4 on Devuan. 2560x1440 monitor, HiDpi disabled, custom scaling (HiDpi text is too big)

PLChris, 2022-12-09:
> This should be added to a 'Known Issues' list for downloaders to see when then install v0.20

Can you make a proposal for the text to be included?

kzembower, 2022-12-21:
Please see [this thread](https://forum.freecadweb.org/viewtopic.php?f=35&t=74615), which I believe is related to this issue.

-Kevin

adrianinsaval, 2022-12-21:
something similar was once reported on archlinux too: https://bugs.archlinux.org/task/73944
as with the other examples, the package from the repo has the issue but a local compilation works fine. I cannot reproduce. What the hell is going on here...

luzpaz, 2023-04-14:
Default Debian and Ubuntu builds of FreeCAD continue to be broken due to lack of maintainership of 3rd party dependencies; slow to integrate more up-to-date versions of said dependencies; and lack of maintainership of FreeCAD builds of said distroa. 

At this time AppImage, flatpack, and Snap packages are recommended as a substitution. 

We are looking for maintainers for the PPAs (stable and development branches), and maintainers in the Debian/Ubuntu ecosystem to help us.

adrianinsaval, 2023-04-14:
I don't think there is direct correlation here to that @luzpaz since we have reports that building locally following the exact same procedure and same dependencies as the packaging results in a working build.

luzpaz, 2023-11-30:
Sorry about misdiagnosing this ticket. Renamed it to Qt related.  
But is this a X/Wayland issue? is it related to #9953 ? 

Please advise.

WandererFan, 2023-11-30:
It has been a while, but I seem to recall fussing with Wayland settings back when I was first investigating.  IIRC, the Wayland settings didn't have any effect.

I don't know that it is a Qt version thing either, as we had reports of it working with version xx.x as well as reports of it not working with the same version.

Something happens in the dpkg, dbuild, etc process that doesn't happen in a regular build of the same source.  

#9953 is not the same problem.  That one is due to the design of the extension functions.

BlueNalgene, 2023-12-02:
I ran into this issue myself using TechDraw, and I was pointed to this issue by @WandererFan [forum link](https://forum.freecad.org/viewtopic.php?p=722761).  The AppImage workaround solved my short term problem, but I want to add my +1 information in case it helps the devs.

FreeCAD from PPA:

```
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 22f524c0c755003c12234fab1bad6a9f696b661a
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

FreeCAD-daily from PPA:
```
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.
Build type: Release
Branch: unknown
Hash: 1faf86c3da12c1712e1d5ec015721d9aa02f8672
Python 3.10.4, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

FreeCAD AppImage (the one that I used to workaround my specific problem):
```
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```

> This issue is not present on latest Snap edge/nightly package. @WandererFan suspects it has to do with Qt5.15.2 since Snap is using Qt5.15.3

My updated problems with rendering occurred on 5.15.3 instead of 5.15.2.  My machine currently has 5.15.3 installed for other applications:
```
$ qtdiag
...... blah blah ....
Qt 5.15.3 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 11.3.0) on "xcb"
...... blah blah ....
```
The AppImage is currently at 5.15.8. Let me know is more information I can offer.  I would love to have the PPA working properly as a lazy end-user.

blazerzar, 2024-02-11:
I observed the same behaviour today while using TechDraw. I am just starting with FreeCAD, so at first I thought I did not know how to properly create lines. After reading the forum, I learned that vertices are not being rendered for me. Using a different version (AppImage) I also noticed that lines were too thin before. When going back I saw that changing the thickness does not work at all.

My main installation where this bug is observed is downloaded from the Arch repository using `pacman`:

```
OS: EndeavourOS (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

Everything works after using the AppImage version on the same machine:

```
OS: EndeavourOS (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

I fully updated EndeavourOS 1 or 2 days ago as well.

ion201, 2024-03-03:
On arch linux I tried to do a local build in a clean chroot and reproduced the same issue as I got with the version from the package manager.

So I built it again, but this time without the clean chroot and that version works properly. Exact same PKGBUILD, only difference was clean chroot build vs "normal" system build. Seems like there's some unaccounted for configuration change which is consistent for both ubuntu and arch linux, so I copied the cmake configuration logs from both runs and compared them. The only difference is noted below. For posterity, the clean chroot build configuration log is here - [freecad_build_config_chroot.txt](https://github.com/FreeCAD/FreeCAD/files/14471470/freecad_build_config_chroot.txt)

```
# Local build, no chroot:
-- CMAKE_CXX_FLAGS:             -Wall -Wextra -Wno-write-strings -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -ffat-lto-objects -fPIC -w
# Clean chroot build
-- CMAKE_CXX_FLAGS:             -Wall -Wextra -Wno-write-strings -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -g -ffile-prefix-map=/build/freecad/src=/usr/src/debug/freecad -flto=auto -ffat-lto-objects -fPIC -w
```

So the only difference between a clean chroot build and my "local" build is the addition of `-g -ffile-prefix-map=/build/freecad/src=/usr/src/debug/freecad -flto=auto` in the CXX compiler flags. Looks like freecad is forcing the addition of debug build flags when built in a clean arch chroot? No idea why that would happen, but maybe someone more familiar with freecad's build proc can make use of that info.

Rahix, 2024-03-05:
LTO (Link time optimization) is a more aggressive optimization strategy where optimizations are performed across compilation units instead of only performing them inside each unit.

If LTO causes bad behavior, this indicates a bug somewhere in the code.  Most likely some subtle undefined behavior that is hard to catch...

What I am trying to say is that disabling LTO is a bandaid fix that might seem to resolve the problem at hand, but there is a deeper underlying issue here.

vladtcvs, 2024-08-17:
I have no vertices if I use FreeCAD 0.21.2 from debian repo or from snap, but if I use FreeCAD from downloaded appimage (with specifying `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libdrm_amdgpu.so.1`, otherwise it crashes), it displays vertices.

herrdeh, 2024-08-21:
Problem exists at ubuntu 22.04 with PPA versions:

* 0.21.2.33771 (Git)
* 0.22.0dev.38115 (Git)

WandererFan, 2024-10-19:
another report of same issue: #13471

herrdeh, 2025-01-18:
This issue is still existing in 

```
ubuntu 22.04.5 LTS
Version: 1.1.0dev.39806 (Git)
Hash: 7829fcc4888bce067fe002a4d857a4953d2482fc
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1

```

Please do not close, a resolve is essential for PPA builds to be usable.

Thanks a lot in advance, 
Wolf

herrdeh, 2025-05-25:
It still exists in:

OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.41841 (Git)

[image]

jgonggrijp, 2025-11-02:
Seeing this with 1.0.2 packaged by openSUSE Tumbleweed:

OS: openSUSE Tumbleweed (KDE/plasma6/xcb)
Architecture: x86_64
Version: 1.0.2.Unknown
Build type: Release
Python 3.13.7, Qt 6.10.0, Coin 4.0.3, Vtk 9.5.2, OCC 7.9.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.43

But not with this version from Flathub (thankfully):

OS: KDE Flatpak runtime (KDE/plasma6/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 256fc7e)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

luzpaz, 2025-11-02:
Any chance you can test on latest dev build ? https://github.com/FreeCAD/FreeCAD/releases

herrdeh, 2025-11-04:
Here:

OS: Ubuntu 24.04.3 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.20251102 (Git shallow) AppImage
Build date: 2025/11/02 16:03:37
Hash: 45ca23d074f39a9b586c8320f1e7cd20ca466fef

vertices are visible and the tool is usable. Thank you for your work!

But I notice that sometimes, vertices (edges as well) are not selected by a single, but by a double click. Rarely, they are not selected at all. I could not find regularity in this behaviour, but it may be kept under observation.

luzpaz, 2025-11-24:
@herrdeh can you add your full [About](https://wiki.freecad.org/About) info so we can compare with other version that are reporting issues ? Thanks!

rabfulton, 2026-06-03:
I just tested pdf export in techdraw which was previously broken for me. Manjaro (arch derivative), X11, Freecad 1.1 and the issue has been fixed for me.

luzpaz, 2026-09-20:
So essentially this is an x11/Wayland issue?

closed 2026-11-04 by chennes