FreeCAD/FreeCAD #6699: [Feature Request] Shortcuts for Draft snaps

FEA-eng, 2022-04-01:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=67658

### Subproject(s) affected?

Draft

### Idea description

Currently, Draft snaps are visible in Tools --> Customize --> Keyboard but shortcuts can't be assigned to them. Shortcuts for these tools could be very useful.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2022-10-31:
With https://github.com/FreeCAD/FreeCAD/pull/7656 shortcuts can be assigned to Draft snaps. But using them during a command only works if the input boxes do not have the focus as they 'catch' the so-called in-command shortcuts (while creating a [Draft_Wire](https://wiki.freecadweb.org/Draft_Wire) in the 3D view the X input box will have the focus).

Roy-043, 2023-02-10:
Let's keep this open. The in-command shortcuts cannot be abandoned obviously but conflict with the 'normal' shortcuts. It is a difficult problem to solve.

This is a reasonable FR, but on the other hand the issue is hardly a deal breaker. I have worked with a 2D CAD program with fixed snap setting without much issue.

kaiwas, 2025-01-25:
Considering that the problem has no solution - is it possible to display some information message? For example, "The key combination is unavailable - press the on-screen buttons!".

It is not right when something works in one place, but not in another. And the user does not understand at all what is happening.