FreeCAD/FreeCAD #6704: Changing Locale should let user know in the UI that a restart is recommended

luzpaz, 2022-04-02:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

n/a

### Subproject(s) affected?

Core

### Idea description

[image]

Just like the option below says (needs restart), chaning a locale **fully** requires a restart. But the UI as of now doesn't let user know this. 

Recommendations: 
we could: 
* add a visual indication like a QLabel saying `(restart recommended)`
* or similar to the Addon Manager, query the user that a restart is recommended. Do they want to do that now or later?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-02:
The fact is that if you change your locale and don't restart, there are certain aspects of FreeCAD that still won't respect the locale change. This is why I'm proposing the top to options

luzpaz, 2022-04-03:
BTW, QtDesigner is hard to use. I tried to break the layout of the .ui and add a Qlabel and jeez it was not cooperating :disappointed:

luzpaz, 2022-04-03:
The more I think of this, the more it merits a dialog asking if the user wants to restart now or later.
CC @donovaly

donovaly, 2022-04-04:
@OpenBrain, is a restart necessary? For me this is not necessary and If a restart is necessary, I think this would be a bug, right?

luzpaz, 2022-09-26:
Restart is necessary see https://github.com/shaise/FreeCAD_FastenersWB/issues/180#issuecomment-1257932236

riqueenz, 2022-09-26:
I agree with @luzpaz. This dialog warning to restart FreeCAD is important.

luzpaz, 2022-09-26:
Added forum thread discussion: https://forum.freecadweb.org/viewtopic.php?f=10&t=72141

adrianinsaval, 2022-09-26:
> The more I think of this, the more it merits a dialog asking if the user wants to restart now or later. CC @donovaly

I disagree and would ask that if this is done it be given a "don't ask again" checkbox, I understand that new users might need to be told this (since it's just a recommendation do they really need to be told?) but as an experienced user I would find these dialogs annoying. And before you ask, yes, as a helper I do change language often enough to make this a real annoyance.

adrianinsaval, 2022-09-26:
I'm fully against forced restart be it for noobs or powerusers. A warning is ok but please give me the "don't ask again" checkbox or at the very least a parameter I can set manually.

luzpaz, 2022-09-26:
Fair enough, then this feature should also provide a `Don't ask again` sub-feature as well.

donovaly, 2022-09-29:
This: https://forum.freecadweb.org/viewtopic.php?p=629195&sid=8288f4801cafa2e19caca6f276422a62#p629195 is also my opinion. At the moment i don't encounter cases where a restart is necessary. If you encounter them, please report them that e.g. @0penBrain can have a look at.

luzpaz, 2022-10-02:
Maybe the folks from https://github.com/FreeCAD/FreeCAD-translations can help me here? What anamolies do you encounter when you don't restart after you change the language prefrerences in FreeCAD ?  
CC 
@kaktusus  
@FreeCAD-Tools  
@Kuzma30

kaktusus, 2022-10-02:
Hej luzpaz

juz we have already talked about this when solving a certain GUI translation problem.

**When you change the GUI language restart is essential.**

Otherwise not all translations may be loaded, I don't know why but it happens.

To make it funny I read about this somewhere on the wiki, but now I won't tell you in which document I came across this information because it was a very long time ago.

0penBrain, 2022-10-02:
@kaktusus please give precise examples if you can so we're able to analyze the causes. Thx. ;)

kaktusus, 2022-10-08:
Here's the thread where it all started ... 
https://github.com/FreeCAD/FreeCAD-translations/issues/87

For the moment I have not been able to track down any situation in which an anomaly occurred.
However, you must know that I have not dealt with this topic in any particular way.

And since I first noticed a problem many months have passed and many FC releases have been published.

0penBrain, 2022-10-08:
@kaktusus OK. I'm not able to reproduce issue in FreeCAD/FreeCAD-translations#87

kaktusus, 2022-10-08:
That was about a year ago
a lot of things have changed since then ...

I will pay more attention to this topic in the future
but I will not dwell on it in any special way ... ok?

There are many other problems more important to translators. ;-)

kaktusus, 2022-10-08:
you know what ... I'm still wondering if this is affected by using local *.qm files with translations _(placed in the translation directory)_

 as far as he remembers @luzpaz would have to confirm if he used this solution when he encountered the problem.

I most likely used local translations that way at the time.

luzpaz, 2022-10-08:
> you know what ... I'm still wondering if this is affected by using local *.qm files with translations (placed in the translation directory)

No, I can say for sure this was not the issue.

donovaly, 2022-10-14:
So this issue can be closed. If you encounter cases where a restart is necessary, please report them as new issue.

closed 2022-10-14 by donovaly

0penBrain, 2022-10-14:
OK. Found some widget types that don't support hot change of language. They don't really need a restart, but some bring user actions to update (such as close/re-open task panel or workbench). Will be quite a mess to fix though. Will report with more details later.

luzpaz, 2023-07-28:
#9996 is another usecase

kaktusus, 2023-09-13:
> @kaktusus please give precise examples if you can so we're able to analyze the causes. Thx. ;)

@0penBrain are you still interested in this topic?

0penBrain, 2023-09-13:
@kaktusus not sure I'm "interested in". 😄 
But I now the reason why some texts aren't live translated, and I may help someone who'd like to tackle this.
But for sure I have neither time nor willing to fix it all over FC by myself. 😉

luzpaz, 2023-09-14:
> But I know the reason why some texts aren't live translated, and I may help someone who'd like to tackle this.

Maybe we can open a new ticket ?

0penBrain, 2023-09-14:
> Maybe we can open a new ticket ?

Yes, this should at least be a discussion to know if we want to globally tackle this.

luzpaz, 2023-11-12:
Folks, we keep encountering this issue.  

> OK. Found some widget types that don't support hot change of language. They don't really need a restart, but some bring user actions to update (such as close/re-open task panel or workbench). Will be quite a mess to fix though. Will report with more details later.

@0penBrain can you provide a list of widgets that do this ?

0penBrain, 2023-11-12:
Not an exhaustive one.
Those for sure that do are ones from QSint.

Also I remember at some point @PaddleStroke introduced such widgets, but with all recent changes in Sketcher I don't know if they still exist.

luzpaz, 2024-09-09:
#16382 is another usecase of incorrect UI due to locale change.