FreeCAD/FreeCAD #6706: [Bug] Keyboard shortcuts not working in undocked doc window

hjalti-thor, 2022-04-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?t=67606

### Version

0.19 (Release)

### Full version info

```shell
OS: Debian GNU/Linux 11 (bullseye) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0f9259c)
Hash: 0f9259cda103ae1824ac16c68ac9b4a0d54b05fc
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)
```

### Issue description

When menu _View → Document window → Undocked_  is selected, keyboard shortcuts stop working
For example, in a _sketch_, drawing a _polyline_, **M** key does nothing when it should cycle through segment types.

### Anything else?

[Wiki documentation](https://wiki.freecadweb.org/Std_ViewDockUndockFullscreen) 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-15:
Page of interest: https://wiki.freecad.org/Std_ViewDockUndockFullscreen
A workaround is presented in this macro: https://forum.freecadweb.org/viewtopic.php?f=3&t=35568&start=10#p571640

Roy-043, 2024-12-15:
This is still an issue.