FreeCAD/FreeCAD #6707: TechDraw: `PropertyFileIncluded` points to wrong location in snap

luzpaz, 2022-04-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://github.com/ppd/freecad-ppd/issues/27#issuecomment-1086731946

### Version

0.20 (Development)

### Full version info

```shell
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28608 (Git)
Build type: Release
Branch: master
Hash: 66d0c35dc780459be0253a990abe2c2afa8fa814
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
Installed mods: 
  * ToolbarStyle
  * Assembly4 0.11.10
  * stickframe
  * sheetmetal 0.2.49
  * pinger
  * Assembly3 0.11.1
  * Behave-Dark-Colors 0.0.1
  * Curves 0.3.0
  * Ship
  * BIM 2021.12.0
  * A2plus 0.4.55a
```
```

### Subproject(s) affected?

Techdraw

### Issue description

When opening [test(1).FCStd.zip](https://github.com/ppd/freecad-ppd/files/8401205/test.1.FCStd.zip) the Snap build complains:

```
18:39:57  Line Group File: /snap/freecad-ppd/214/usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
18:39:57  Svg Hatch File: /snap/freecad-ppd/107/usr/share/Mod/TechDraw/Patterns/simple.svg is not readable
18:39:57  Pat Hatch File: /snap/freecad-ppd/107/usr/share/Mod/TechDraw/PAT/FCPAT.pat is not readable
```

According to @WandererFan 

> The filenames for the "included" files come from `App::PropertyFileIncluded::getExchangeTempFile()`

and 

> If `PropertyFileIncluded` is looking in the wrong place for the file that is part of the FCStd file, then that seems like a problem.

> If you are sharing an FCStd file with somebody that doesn't have a `/usr/share/freecad/...` (like a Win user) to fall back on, then there will be an issue.

> Assuming TD is using `PropertyFileIncluded` properly (???), then the logic for flatpack/appimage/snap versions needs to be in `PropertyFileIncluded`. TD can't be the only place `PropertyFileIncluded` is used.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-03:
Easier way to reproduce: 

1. Start FC
2. Switch to TechDraw WB
3. Create new document (ctrl+n)
4. Choose 'Insert Default Page'

**Result**: `Template File: /snap/freecad-ppd/107/usr/share/Mod/TechDraw/Templates/A4_LandscapeTD.svg is not readable`

WandererFan, 2022-04-04:
All of the "not found" files use App::Application::getResourceDir to build the extended file name, as in:
    std::string defaultDir = App::Application::getResourceDir() + "Mod/TechDraw/PAT/";

What do the numbers(107, 214) mean in: 
/snap/freecad-ppd/**107**/usr/share/Mod/TechDraw/PAT/FCPAT.pat

WandererFan, 2022-04-04:
getResourceDir returns /snap/freecad-ppd/**${revision_number}**/usr/share/ 
Apparently, it should return  /snap/freecad-ppd/**current**/usr/share.

getHomePath returns /snap/freecad-ppd/${revision_number}/usr/
It may be that this should return /snap/freecad-ppd/current/usr, but I'm can't confirm this.

It looks like FC determines these directories from the path to the executable.   If it is possible to launch the executable from "current" instead of "${revision_number}", that might solve the issue.

luzpaz, 2022-04-05:
So this is a Snap packaging issue? Or should we add logic that ascertains where the `getResourceDir` by querying the system instead of assuming it's the path for the executable?

WandererFan, 2022-04-05:
How this occurs:

- install snap revision 001
- set file location preferences to defaults based on getResourceDir
- defaults will point to install directories of revision 001
- create xxxx.FCStd that uses file properties (templates, hatches, etc) and leave file locations at defaults

- install snap revision 002
- 002 will be installed to a different location from 001
- revision 001 directories may or may not still exist due to snap installation garbage collection
- load xxxx.FCStd which has file properties pointing to revision 001 directories
- if revision 001's directories still exist then no problem
- if revision 001's directories have been deleted, then file properties are invalid

In a stable environment, where new versions are not installed, and old versions are not deleted, this situation will not arise.

This is a problem for a development environment where revisions occur frequently.

WandererFan, 2022-04-05:
> So this is a Snap packaging issue? Or should we add logic that ascertains where the `getResourceDir` by querying the system instead of assuming it's the path for the executable?

We could write code that says "if this file is in a Snap directory and it can't be read, change the file name to point to the current Snap installation and try again".  But when do you stop?  Do you search the whole disk for a likely looking file and use that? 

Maybe we just tell people using the "edge" Snap that file references might go stale and need to be updated manually.

If they are using the stable Snap, they should never run into this situation.

luzpaz, 2022-04-06:
CC @ppd

ppd, 2022-04-06:
In general, there seem to be two different problems, as far as I understand it:

1. References to absolute paths will break if files move between installations, e.g. Windows to AppImage, or Snap to Windows etc. This is unavoidable if you do not use relative paths. To solve this, you'd need path variables, e.g. `${FREECAD_RES}/some/dir/pattern.svg` or similar. This has nothing to do with snaps and is a problem in itself.
2. Snap revision 1 to  Snap revision 2 breaks, too. This is for the same reason as (1), but more unexpected since you're still on the "same" installation

I propose: Introduce a variable $FREECAD_RES or similar that gets expanded when such files are accessed. Construct paths relative to this root variable. Make it so that it can be set from the environment, so the snap could set it to `$SNAP/usr/share...` (whatever). If it's not set, deduce it from the executable path, as is the case presently.

luzpaz, 2022-07-03:
> This has nothing to do with snaps and is a problem in itself.

> I propose: Introduce a variable $FREECAD_RES or similar that gets expanded when such files are accessed. 

So essentially this solution will improve interoperability between sharing FCstd between different users on different plaforms no matter the install methodology.

ppd, 2022-07-05:
@luzpaz Yes, that sounds about right. If the file format allows references to assets outside the FCStd container, they must be relative to some configurable root.

RWL-69, 2024-02-02:
Seeing this issue is interesting as users of appimages are having the same problem as described in [this discussion thread](https://forum.freecad.org/viewtopic.php?t=74668) .  Users of appimages find that random directories are created every time that you start the appimage.  Thus, direct reference to that directory can break hashes or other items.

It would be nice to be able to setup a setting in the configuration to point to a local directory like the option for welding symbols.  I have tried to setup a techdraw directory in .local/share/FreeCAD/Mod and it stops the TechDraw workbench from appearing in the drop down workbench menu.

So, this bug affects not just snap users but appimage users as well.

luzpaz, 2024-02-10:
@adrianinsaval JFYI :point_up_2:

RWL-69, 2024-12-09:
Just checked with Version 1.0 appimage, I can set directories in the Preferences, TechDraw, General for files. 

Did a quick test with 1.0 and a file created in TechDraw was created and saved.  FreeCAD was closed and reopened and the template for the drawing was still opened and populated as expected.  Would need to do further testing but in my opinion, this can be closed.

luzpaz, 2025-01-15:
@RWL-69 can you report back any progress ?

RWL-69, 2025-01-16:
Hello Luzpaz,  

I stated in December that in my opinion, this issue could be closed.  Is that not enough?  My testing at that time showed that users can define their own directories which fixes the issue from appimages for TechDraw.  That is with the 1.0 appimage.

luzpaz, 2025-01-16:
Thanks for confirming. Closing.

closed 2025-01-16 by luzpaz