FreeCAD/FreeCAD #6719: [Feature Request] Let User set the order of drilling (helix) targets.

sliptonic, 2022-04-06:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=67775&p=586671#p586671

### Idea description

The user may want to override the default order of drilling targets.  He should be able to sort the list as desired.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

J-Dunn, 2022-04-07:
I've often been frustrated by the lack of control on this feature too. This can be inconvenient when automatic routing is unsuitable in particular with avoiding clamps, for example. 

It would be nice if the order could be set in the base geometry tab of the UI, if that widget can allow that or could be modified to allow user reordering.

J-Dunn, 2022-04-07:
Poking around in UI files it seems this should be possible but Qt widget is behaving oddly and it seems to lose the list entry on drop.  I guess others have looked at this and decided it was a bit flakey.

Apart from losing the first entry on dropping , the list reorders well so presumably this could be used to let the user reorder the holes as desired, with some simple coding to adjust path order to the order of the list. 

[quote]~/FreeCADR/src/Mod/Path/Gui/Resources/panels/PageBaseHoleGeometryEdit.ui

dragDropMode = internalMove
defaultDropAction = TargetMoveAction
dragEnabled=True
acceptDrops=True
dragDropOverwriteMode=false[/quote]

sliptonic, 2022-04-07:
Of course it's possible.  That's why I added the feature request issue.  It isn't even difficult now that the helix and drilling have been refactored to generators.  But it won't happen for 0.20.  If you want to work on a PR for 0.21, please do.

J-Dunn, 2022-04-09:
This allegedly works for QTableView but the vertical header needs to be visible ( not really a problem ) and a called is needed to setSectionsMoveable() 

https://bugreports.qt.io/browse/QTBUG-13873?page=com.atlassian.jira.plugin.system.issuetabpanels%3Aall-tabpanel

However, that does not seem to be available from python :(  Or at least I can't find what it's called. 

The ui may have to be reorganised around QListWidget like the operations reordering. 

```
    def setFields(self, obj):
        """setFields(obj) ... fill form with values from obj"""
        PathLog.track()
        self.form.baseList.blockSignals(False)
        self.form.baseList.dragEnabled=True 
        self.form.baseList.verticalHeaderVisible=True
        self.form.baseList.verticalHeaderStretchLastSection=True
#        self.form.baseList.verticalHeader.sectionsMovable=True
        self.form.baseList.verticalHeaderSectionsMovable=True

```