FreeCAD/FreeCAD #6727: [Bug] Missing B-Spline in STEP file after Part Offset2D

mpetrasinovic, 2022-04-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=67776

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.27422 (Git)
Build type: Release
Branch: master
Hash: d938733eaf2c2ce7cb18d1cbb56147185c473530
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Part

### Issue description

After applying Part Offset2D to a sketch (or Wire) that has B-Spline it is computed and shown like it is supposed to but when we export this geometry to STEP all B-Spline offsets are ignored.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-09:
CC @AjinkyaDahale do you mind weighing-in?

AjinkyaDahale, 2022-04-10:
I'll take a look when I can. @PaddleStroke you are also working on offsets for sketches right? Anything that strikes a bell?

mpetrasinovic, 2022-04-10:
I want to add that spline from the sketch is exported, but one from the offset is ignored.

The most promising workaround (that maybe can be implemented behind the scene) is to use the same underlying functions that Draft Shape2DView uses but to keep geometry at the exact placement. After this function, everything is exported as expected.

PaddleStroke, 2022-04-10:
Unfortunately I don't know how offset 2D tool works in details. I know how it gets the offset geometries from OCC but that's about it. And this part is working OK because you can see the offset bspline before exporting. 
So my guess is that the issue is at the export function. I would guess that the handling of the offset 2D object by the export has a problem with bspline (or they may not even be coded).
Because in Part, the offset2D is an object, it's not actually a bundle of the offset edges. Which enables to change the offset easily. So I guess the export has to get the offset edges from the offset2D object.
My plate is quite full at the moment so I can't have a deeper look in it.

wwmayer, 2022-04-11:
It's not only a problem with B-Spline but also ellipses, parabolas and hyperbolas are affected.

When looking at the output shape of the 2D offset then in most cases it's a wire of several edges. Depending on the geometry of the input shape the edges of the output shape can be lines, circles, B-Splines and sometimes the curve type "offset curve".

It looks like this curve type is causing problems and for me a STEP file with an exported offset curve can't even be loaded. What works fine is to export the offset curve as IGES because here the translator obviously converts the offset curve into one or multiple B-Splines.

mpetrasinovic, 2022-04-11:
I have posted on the forum model that has problems also with IGES export and import  (https://forum.freecadweb.org/download/file.php?id=185333&sid=b4b162c8dcc60fa209123547bbcbafd7). I tried several times to use IGES as a workaround but it transforms solid to faces while exporting. This makes new problems for usability.

wwmayer, 2022-04-11:
Here is a function to convert the offset curve of a wire with B-Splines: 3d249e0e67

wwmayer, 2022-04-11:
a13e251ad4 uses the newly added function to replace offset curves. Please check if it solves the issue.

mpetrasinovic, 2022-04-11:
I tried to compile FreeCAD again and this time I have got:

[image]

mpetrasinovic, 2022-04-11:
I guess this is now the limit of the Offset function, I have changed the spline to the following one and now everything works, the model is exported to STEP and imported back without any problem.

[image]

mpetrasinovic, 2022-04-11:
I just checked, offsetting ellipse also worked as expected.

wwmayer, 2022-04-11:
> I have complied master and now if sketch or wire has BSpline I get makeOffset2D: result of offsetting is null! Input error.

I think that's an issue that must have happened before the change, too. 

> I tried to compile FreeCAD again and this time I have got:

Do you have a test file? What could possibly happen is that the edges of the wire are not in the expected order and thus if a new edge isn't adjacent to an edge of the wire it will be rejected.

> I guess this is now the limit of the Offset function

There should be no limitation compared to the old behaviour

donovaly, 2022-04-12:
I gave it a try and for me the issue is resolved.

mpetrasinovic, 2022-04-12:
> I think that's an issue that must have happened before the change, too.

Yes, I also think so.

> Do you have a test file? What could possibly happen is that the edges of the wire are not in the expected order and thus if a new edge isn't adjacent to an edge of the wire it will be rejected.

Maybe, I will try to find that file. But I will tomorrow have the same computer.

> There should be no limitation compared to the old behaviour

I do not think there is a limitation compared to the old behavior.

wwmayer, 2022-04-12:
> I do not think there is a limitation compared to the old behavior.

But in your second picture an edge of the offset wire is missing. However, it's also possible that it was already missing before doing the replacement.

I have tested the offset function a bit further by using an input wire where the edges are added in arbitrary order and the replacement still worked fine.

mpetrasinovic, 2022-04-14:
I tested some sketches and found this one that I can't offset.

(added zip extension in order to be able to upload it)

[2d-offset-sketch-bug.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/8489980/2d-offset-sketch-bug.FCStd.zip)

mpetrasinovic, 2022-04-14:
I think that there are two coincident control points, which is probably why offset is not working.

> But in your second picture an edge of the offset wire is missing.

I can't reproduce that again.

donovaly, 2022-04-15:
> I tested some sketches and found this one that I can't offset.
> [2d-offset-sketch-bug.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/8489980/2d-offset-sketch-bug.FCStd.zip)

Confirmed using recent master.

wwmayer, 2022-04-16:
I can confirm the failure, too. But fortunately, it's not a regression of my fix. The OCC algorithm already fails to create an initial wire from the sketch.

mpetrasinovic, 2022-04-16:
The reported bug is then solved. The latest reported problem is a different bug. This should be closed?

donovaly, 2022-04-16:
Closing as fixed then

closed 2022-04-16 by donovaly