FreeCAD/FreeCAD #6731: [Bug] freecad crashes opening IFC file, which was generated by itself

voukka, 2022-04-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/

### Version

0.19 (Release)

### Full version info

```shell
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git)
Build type: Release
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: C/Default (C)
```

### Subproject(s) affected?

Arch

### Issue description

Opening .ifc file crashes FreeCAD every single time. IFC file was generated from the attached .FCStd file, which I got going through the [BIM tutorial](https://wiki.freecad.org/Manual:BIM_modeling).

Zip archive with the original .FCStd and .ifc files [Archive.zip](https://github.com/FreeCAD/FreeCAD/files/8459131/Archive.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-10:
Crash output? If you dont see any, please start FreeCAD from the CLI

luzpaz, 2022-04-10:
I can open the FC file just fine. the IFC file I can't open because I don't have IfcOpenShell installed for the Snap package. 

`11:07:55  IfcOpenShell was not found on this system. IFC support is disabled.
Visit https://wiki.freecadweb.org/IfcOpenShell to learn about installing it.`

[code]
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28671 (Git)
Build type: Release
Branch: master
Hash: aea110ba15897661a59f04ec52c701a00daea430
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
Installed mods: 
  * ToolbarStyle
  * Assembly4 0.11.10
  * stickframe
  * sheetmetal 0.2.49
  * pinger
  * Assembly3 0.11.3
  * Behave-Dark-Colors 0.0.1
  * Curves 0.4.1
  * Ship
  * BIM 2021.12.0
  * A2plus 0.4.55b
[/code]

luzpaz, 2022-04-10:
CC @yorikvanhavre (should probably be moved to the BIM WB repo)

donovaly, 2022-04-11:
All verified crashes and regressions can be put to the 0.20 milestone

donovaly, 2022-04-11:
I see now that the crash was not verified.

@voukka , please open a forum thread. This is our rule. When others can reproduce the issue it can be reported.

I am closing this now. Please reopen when others could verify the crash.

closed 2022-04-11 by donovaly