FreeCAD/FreeCAD #6733: [Bug] DraftView - line style property not working

FEA-eng, 2022-04-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=35&t=67928

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.28611 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

DraftView objects have the Line Style property but it doesn't work - there's no drop-down list for it and typing in the style name (e.g. "Dashed") does nothing. Such property can be useful to turn draft objects to annotations.

The style of original Draft objects being turned into views is also not considered so there is no way to change the style of lines created by the DraftView tool.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-04-12:
This has to be fixed on the Draft/Arch side, I believe.

WandererFan, 2022-04-13:
If you take the text of DraftView's Symbol property and give it to Inkscape you will get what looks like solid lines. Inkscape will tell you that in fact you have a line with a custom dash property, which is correct. 

The issue is this code "stroke-dasharray:0.09,0.05" in the style attribute.  This code creates a dashed line made up of very small dashes separated by very small gaps.  The result appears as a continuous line.

yorikvanhavre, 2022-05-12:
The LineStyle property is working. The problem is that SVG is a bit delicate with line patterns... It gets scaled if placed inside a scaled group. The Draft get_svg() function normally takes care of this, at https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/draftfunctions/svg.py#L52 but it still depends on what defaults are set under Edit->Preferences->Draft->Visual settings. So far I've been unable to find values that work for everyone. Some values work when you are working on a model of a building, but not when working on a screw model...

Not really sure how to handle this... Maybe the whole system needs to be redone with something more accurate, like "I want 3mm dashes with a 1mm spacing" and FreeCAD would do the whole calculation up to the printed sheet...

Roy-043, 2022-07-21:
SVG patterns refer to [Draft Patterns](https://wiki.freecadweb.org/Draft_Pattern). Why do you think these settings are related?

paullee0, 2022-07-21:
Just wild guess, the only two inputs with figures looks like multiplier ?

luzpaz, 2023-04-13:
@yorikvanhavre do you mind revisiting this ?

yorikvanhavre, 2023-05-08:
On it!

yorikvanhavre, 2023-05-10:
Above commit implements more sensible default values that give good results at all scales:

[image]

paullee0, 2023-05-10:
Looks good :)

PR #9530 (2023-05-15): Draft: Better svg linestyle settings - fixes #6733
Better defaults for dashed, dotted and dashdotted lines in TechDraw DraftView objects
fixes #6733

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
commit 3c685c55 (2023-05-15): Draft: Better svg linestyle settings - fixes #6733
commit a892de5f (2023-05-15): Merge pull request #9530 from yorikvanhavre/draft-svg-dashes

Draft: Better svg linestyle settings - fixes #6733
closed 2023-05-15 by yorikvanhavre