FreeCAD/FreeCAD #6735: [Bug] [regression]

spanner888, 2022-04-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=67882

### Version

0.20 (Development)

### Full version info

```shell
OS: Ubuntu 21.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28657 (Git)
Build type: Debug
Branch: master
Hash: fb1f4433c6bdafc5ec58bea0e1ea799f302c0bf1
Python 3.9.5, Qt 5.15.2, Coin 4.0.0, OCC 7.5.2
Locale: French/France (fr_FR) [ OS: French/France (fr_FR) ]
```

### Subproject(s) affected?

Core

### Issue description

```
import FreeCAD

q = FreeCAD.Units.Quantity 

kFactor = q('1400.0 N/mm^2')
print("Kf : ", kFactor, kFactor.Unit)
```

Works fine in FC 0.19 and gives output:

`13:33:53  Kf :  1400000.0 kg/(mm*s^2) Unit: kg/(mm*s^2) (-1,1,-2,0,0,0,0,0) [Pressure]`

But in Fc 0.20, gives error:

```
# 13:48:46  Traceback (most recent call last):
#  File "/home/spanner888/.local/share/FreeCAD/Macro/units@001.FCMacro", line 5, in <module>
#    kFactor = q('1400.0 N/mm^2')
#<class 'ValueError'>: syntax error
```

Confirmed by [openBrain](https://forum.freecadweb.org/memberlist.php?mode=viewprofile&u=22265)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-12:
CC 0penBrain

luzpaz, 2022-04-12:
> Confirmed on below version

```
OS: Ubuntu 21.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28657 (Git)
Build type: Debug
Branch: master
Hash: fb1f4433c6bdafc5ec58bea0e1ea799f302c0bf1
Python 3.9.5, Qt 5.15.2, Coin 4.0.0, OCC 7.5.2
Locale: French/France (fr_FR) [ OS: French/France (fr_FR) ]
```

> You can workaround it by adding a space on either side of the division bar (eventually both).
> From quick tests, the problematic character sequence is specifically 'N/mm'.

0penBrain, 2022-04-12:
BTW I don't see purpose of this commit. This isn't adding atomic unit but composed unit, which IMO are indeed prone to break the system.

PR #6737 (2022-04-13): [BUGFIX]Core: partially revert 4ab37fc that introduced composed units
 Parser shall only deal with atomic units or it breaks other composed units
 Fixes #6735
commit 78e11740 (2022-04-13): Core: partially revert 4ab37fc that introduced composed units

 Parser shall only deal with atomic units or it breaks other composed units
 Fixes #6735
closed 2022-04-13 by donovaly