FreeCAD/FreeCAD #6739: [Bug] Document recompute does not honor configured Sketcher default Solver

mnesarco, 2022-04-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=22&t=67896

### Version

0.20 (Development)

### Full version info

```shell
OS: Linux Mint 19.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Issue description

After configuring Sketcher default solver and restarted Freecad, recompute still uses DogLeg solver, until you manually open the sketch editor to allow the UI code to read the configured value.

1. Configure non DogLeg solver
[image]

2. Restart FreeCAD
3. Open the file
4. Recompute (As you can see, DogLeg is still used)
[image]
5. Open the sketch (opening the sketch editor, loads the configured solver)
[image]
6. Close the sketch editor
7. Recompute (As you can see, now the configured Solver is used)
[image]

### Anything else?

This Happens because defaultSolver parameter is only loaded by the UI and the DogLeg Solver is hardcoded in Sketch constructor: https://github.com/FreeCAD/FreeCAD/blob/a13e251ad45c3562875e6bcc8e1c7e84882a4d52/src/Mod/Sketcher/App/Sketch.cpp#L80

So without UI loading, the Sketch object ignore the configured value.

Currently only GUI code knows about the configuration and loads it: 
- https://github.com/FreeCAD/FreeCAD/blob/a13e251ad45c3562875e6bcc8e1c7e84882a4d52/src/Mod/Sketcher/Gui/TaskSketcherSolverAdvanced.cpp#L97
- https://github.com/FreeCAD/FreeCAD/blob/a13e251ad45c3562875e6bcc8e1c7e84882a4d52/src/Mod/Sketcher/Gui/TaskSketcherSolverAdvanced.cpp#L401

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-14:
This ticket is formatted really well! Thanks for the effort and research!!!

luzpaz, 2022-04-14:
@mnesarco Question: are you positive this issues is also accurate in FreeCAD master (since you're using LinkDaily as the basis for this bug report)?

mnesarco, 2022-04-14:
@luzpaz yes, the code I am referencing is from FreeCAD master. I also have a clone of master in my local env.

mnesarco, 2024-10-30:
Nobody cares about this issue. Closing. If someone else have this problem in 1.0, please open a new Issue.

luzpaz, 2024-10-30:
@mnesarco apologies. Can you still reproduce this issue in 1.0 or 1.1-dev ?

mnesarco, 2024-10-30:
Hello luzpaz, yes, nothing have changed since then. 

```
OS: Ubuntu 24.10 (xcb)
Architecture: x86_64
Version: 1.1.0dev.39096 (Git)
Build type: Release
Branch: main
Hash: 4f323f9580334de25192d2bf1f272206ad2d3f55
Python 3.12.7, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1

```

This is a very low priority bug as apparently nobody changes the default solver (it is an undocumented parameter)

mnesarco, 2024-10-30:
I wonder why Solver is not a Property of the Sketch. It feels more natural to set the solver on a per sketch case, so it should not depend on User preferences. If you share a project with someone and your sketch depends on a specific solver, it will behave differently if the other user have a different solver in preferences.

luzpaz, 2024-10-30:
> I wonder why Solver is not a Property of the Sketch. It feels more natural to set the solver on a per sketch case, so it should not depend on User preferences. If you share a project with someone [...]

Maybe we need to open another ticket as a FR for this. And link to this one

DavidLister, 2025-03-18:
I still see this issue in the latest release, I think it would be worth re-opening.

One use-case that this makes more challenging is linkage design. Sketches are great for studying and designing linkage motions, but this requires dragging points along the trajectory. Unfortunately dogleg is not well suited for this, and the advanced solver options don't change the default solver for solving while dragging a point.

It may be niche feature, but I think it would help a non-empty subset of users a fair bit!

FEA-eng, 2025-11-18:
> 3\. Open the file

A specific file (nothing is attached)? Can you confirm if this issue is still relevant?

luzpaz, 2025-11-18:
> Maybe we need to open another ticket as a FR for this. And link to this one

If this is still valid, this should be done as well.

closed 2026-09-11 by maxwxyz