FreeCAD/FreeCAD #6741: [Bug] PD primitives cannot be copied to another body

donovaly, 2022-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=588092#p588092

### Version

0.19 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

PartDesign

### Issue description

PartDesign primitives, no matter if substractive or additive, cannot be copied to an other body.

- copy an additive box from one body and paste it to a second, existing body
- as result the copied boy stays in the tree at the same level as the body
- now try to drag them into the desired body

result: this is impossible. However for others objects like sketches or datum planes, this works without problems

See the forum thread for a testcase.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-12-05:
I don't understand. Today I thought I ran into this bug. But now that I try to reproduce it in a simple case I can't.

- Create body, sketch, pad
- Create 2nd body.
- Select the pad. CtrlC. Unselect the XY-plane from the 'Object selection' dialog.
- CtrlV will paste the feature as root object
- Dragging it to the new body is possible.

So perhaps this issue should be closed since it appears not to be impossible to drag into another body. And there is already 6985 that describe the issue that the pasted feature is at root.

@maxwxyz Could you double check and close if you confirm?

macdroid53, 2024-12-05:
IMO, the bug here is that FC allows the user to paste a feature of a Body in the top level of the tree...outside a Body.

PaddleStroke, 2024-12-05:
I agree, but this is what you describe in 6985, so this ticket here can be closed

macdroid53, 2024-12-05:
Yes, agreed.

maxwxyz, 2024-12-05:
Closing then

closed 2024-12-05 by maxwxyz