FreeCAD/FreeCAD #6744: [Bug] Spreadsheet merge/split cells bug

m1cha1, 2022-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=67969

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.28671 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Polish/Poland (pl_PL)

[/code]
```

### Subproject(s) affected?

Spreadsheet

### Issue description

Steps to reproduce:
Problem with merging
1. Merge cells A1 & B1
2. Merge A1 & B1 & C1 - it's not working

Problem with splitting
1. Merge cells A1 & B1
2. Merge cells C1 & D1
3. Select the range from A1 to D1 and split it - it's working partially, only C1 and D1 are splitted

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

0penBrain, 2022-04-13:
As a whole, PR #6748 should resolve this ticket.
There are possible further improvements as discussed in the forum thread, but that should IMO be traced as other feature improvement tickets.

closed 2022-04-15 by chennes

wwmayer, 2022-04-16:
Are you sure the issues are solved? Especially none of the issues described by the OP in the forum are solved for me. It's even worse now. While in an older version the splitting worked for C1+D1 it now doesn't work at all any more because the command is disabled.

commit 2a002e3d (2022-04-15): Sheet: Warn user when cell merge is not done because already merge cells in the range

 Partially fixes #6744
commit 384f56fe (2022-04-15): Sheet: enable Split action only if only one merged cell is selected

 Both main menu and context menu
 Partially fixes #6744
commit 4f75f86a (2022-04-15): Sheet: update commands status when selection is changed

 Notice it doesn't cover the case of the context menu actions
 Partially fixes #6744