FreeCAD/FreeCAD #6746: [Bug] Draft breaks the model, braking following features

YosypC, 2022-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=67941

### Version

0.20 (Development)

### Full version info

```shell
My personal:

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/Europe (en_150)
```

### Subproject(s) affected?

PartDesign

### Issue description

A draft breaks the model (although not visually), bringing problems in later PartDesign features.

### Anything else?

https://github.com/realthunder/FreeCAD/issues/333
https://github.com/realthunder/FreeCAD/issues/334

Everything (including my model) is in there. The discussion is for Linkstage3's branch, but applies to Master too.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-14:
Per @kisolre 
> Draft creates a broken model. Pad after that is not independent. It is created on top of Draft, which is already broken so Pad is also broken. Draft shows multiple errors because of `Refine=True`. There are some errors even without refine. Also when I open the file in master there is no source for the latest Pad.

adrianinsaval, 2023-01-02:
please attach a file created in master branch showcasing the issue

adrianinsaval, 2023-02-07:
no response so I suggest closing, this is not a place for dealing with linkstage bugs, if the issue is indeed reproducible in latest master a file fully created in master is required

YosypC, 2023-02-24:
> no response so I suggest closing, this is not a place for dealing with linkstage bugs, if the issue is indeed reproducible in latest master a file fully created in master is required

Unfortunately I have no time to dedicate in recreating the bug fully in vanilla FC, therefore I'll close the issue. I'm sorry if I caused any annoyance.

luzpaz, 2023-02-24:
@kisolre by chance, can you reproduce ?

kisolre, 2023-02-25:
> @kisolre by chance, can you reproduce ?

 I already mentioned since there is no new file so there is now new way to properly test. Cant recompute original file as a whole because a lot of Linkstage3 features used. So unless a new file is supplied nothing new to test. Also those extreme drafts would probably still be problematic in some cases.

luzpaz, 2023-02-26:
Thanks!

Closing ticket

closed 2023-02-26 by luzpaz