FreeCAD/FreeCAD #6757: [Feature Request] QMdiArea supports two modes: tabbed and sub-window. Implement a way to toggle between them

luzpaz, 2022-04-16:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=588960#p588960

### Subproject(s) affected?

Core

### Idea description

Per wmayer:  
> The class [QMdiArea](https://doc.qt.io/qt-5/qmdiarea.html#ViewMode-enum) supports two modes: tabbed and sub-window. So, you can open a ticket for a feature request for 0.21 to implement a way to toggle between the two modes.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-13:
@luzpaz, when user clicks _Windows_ → _Tile_ or _Cascade_, _FreeCAD_ shows subwindows. What kind of behavior did the user request? So far I concluded that he wanted the tabs to disappear to get more space and less confusion. The tabs indicate view, the TreeView "shows all user defined objects that are part of a FreeCAD document".

luzpaz, 2024-09-14:
@xtemp09 honestly i'm not sure. Do you mind bumping  that forum thread with that question ?