FreeCAD/FreeCAD #6759: [Feature Request] Dowel/center of gravity symbol

FEA-eng, 2022-04-16:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=35&t=66789

### Subproject(s) affected?

Techdraw

### Idea description

A symbol common for dowel pin and center of gravity could be useful in TechDraw.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-06:
@FEA-eng what is the proposed way to implement this ? (to help onboard any new devs that would be interested in contributing)

pierreporte, 2023-11-06:
> But, the ideal and full solution would be to make a tool that could add this symbol to the view in the location of the center of gravity of the part (as discussed in the forum thread) with an option to move it freely if needed.

It should be possible to snap these to the centers of circles. Similar symbols are often used to code for holes types when there is a lot of them. You put the symbols at the centers and write the specification elsewhere in the drawing.

It would also be great to be able to position them with precision. Say you want to dimension your part from a certain point: you put the symbol on the drawing and put dimensions that could act as constrains like in sketch. This is also widely used in the industry. (There are also lines that could behaves in a similar fashion, like gauge lines.)

pierreporte, 2023-12-05:
PR #11655 can be the first step to solve this issue. By allowing to position a point precisely in a view, the symbol could be snapped on that point or the method be reused to set its dimension.

pierreporte, 2023-12-05:
Also it would be useful to have more than the center of gravity symbol. Here are 7 more similar symbols that could be used for various purposes, like hole callouts:

[image]