FreeCAD/FreeCAD #6762: [Bug] Involute gear tool duplicated in the menu

FEA-eng, 2022-04-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=68069

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.28717 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

Steps to reproduce:
1) Create a new body
2) Create involute gear, leave the default settings
3) Pad with a default length of 10 mm
4) Create another body
5) Create another involute gear, change the number of teeth to 13
6) Pad with a default length of 10 mm
7) Activate the first body, click the top face of the gear and create a sketch on it
8) Close the sketch

Now there are two Involute gear options in the menu and FreeCAD has to be restarted to remove this redundant option.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-04-18:
Confirmed. The issue is also in OS:

Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Germany (de_DE)

luzpaz, 2022-04-18:
Confirmed on Linux
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28719 (Git)
Build type: Release
Branch: master
Hash: 139370a20fdb23b900424465bec967c7187ab931
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
Installed mods: 
  * ToolbarStyle
  * Assembly4 0.11.10
  * stickframe
  * sheetmetal 0.2.49
  * pinger
  * Assembly3 0.11.3
  * Behave-Dark-Colors 0.0.1
  * Curves 0.4.2
  * Ship 2022.4.11
  * BIM 2021.12.0
  * A2plus 0.4.56
```
[image]

closed 2022-04-19 by wwmayer