FreeCAD/FreeCAD #6765: [Bug] Overlay Icons for Std Link nearly invisible small

PLChris, 2022-04-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=68061&p=589456#p589444

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.28717 (Git)
Build type: Release
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * DynamicData 2.45.0
  * FeedsAndSpeeds 0.4.0
  * fcgear
  * A2plus 0.4.54b
  * fasteners
  * sheetmetal
  * ExplodedAssembly
  * Curves 0.3.0
[/code]
```

### Issue description

The overlay icons are on a high resolution near to invisible.
Other overlays e.g. indicating non attached or not fully constrained sketches have their normal size.

### Anything else?

Display resolution is 3072x1920 (Retina display)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-18:
Added screenshot to ticket

donovaly, 2022-04-18:
Is this also in FC 0.19?

luzpaz, 2022-04-26:
Thanks wwmayer!

* Are we going to backport?
* Can we close this ticket?

donovaly, 2022-05-01:
No backport since there won't be a further 0.19 release.
Closing as fixed.

> We should update the images in the wiki page: https://wiki.freecad.org/Tree_view#Overlay_icons

Yes, please do so when you find time.

luzpaz, 2022-05-01:
[image]
ref: https://github.com/FreeCAD/FreeCAD/blob/4f9eb9ebcea67bff2a46cfc97ae125472f115a11/src/Gui/Icons/resource.qrc#L230-L233

Are we keeping the xpm files for any reason?

luzpaz, 2022-05-01:
Also the overlay icons in the wiki page look nothing like the ones that were altered in wwmayer's commit:

https://github.com/FreeCAD/FreeCAD/blob/0cf9a778a54cf4aff76863ec1bf4c265b2596ef0/src/Gui/Icons/LinkOverlay.svg
https://github.com/FreeCAD/FreeCAD/blob/0cf9a778a54cf4aff76863ec1bf4c265b2596ef0/src/Gui/Icons/LinkSubElement.svg
https://github.com/FreeCAD/FreeCAD/blob/0cf9a778a54cf4aff76863ec1bf4c265b2596ef0/src/Gui/Icons/LinkArrayOverlay.svg
https://github.com/FreeCAD/FreeCAD/blob/0cf9a778a54cf4aff76863ec1bf4c265b2596ef0/src/Gui/Icons/LinkSubOverlay.svg
https://github.com/FreeCAD/FreeCAD/blob/0cf9a778a54cf4aff76863ec1bf4c265b2596ef0/src/Gui/Icons/LinkElement.svg

I did a quick search for remaining xpm files in the source: (looks like most are overlay icons)
```
$ find -name *.xpm
./src/Gui/Icons/LinkArrayOverlay.xpm
./src/Gui/Icons/button_right.xpm
./src/Gui/Icons/LinkOverlay.xpm
./src/Gui/Icons/button_up.xpm
./src/Gui/Icons/LinkSubElement.xpm
./src/Gui/Icons/freecad.xpm
./src/Gui/Icons/LinkSubOverlay.xpm
./src/Gui/Icons/button_left.xpm
./src/Gui/Icons/button_down.xpm
./src/Gui/Icons/bulb.xpm
```
None of these resemble what is on the wiki page:  
[image]

donovaly, 2022-05-04:
If possible the XPM files should go and we use only SVG files. If you can make such a PR, please do so and this should go in for 0.20.

luzpaz, 2022-05-06:
@donovaly sorry, i know you're busy but there are two issues:
1) I can remove the XPMs but I wouldn't know if there were correctly replaced by their SVGs
2) I haven't heard a response about the pending question: where are the overlay icons used in the treeview (the ones wwmayer updated look nothing like the overlay icons in the wiki)

donovaly, 2022-05-07:
> I can remove the XPMs but I wouldn't know if there were correctly replaced by their SVGs

I would say just compare the SVG with the XPM. If it fit, remove the XPM, if not put it on the list that our icon guys like @bitacovir  can create a proper SVG/fix the SVG

> where are the overlay icons used in the treeview (the ones wwmayer updated look nothing like the overlay icons in the wiki)

I don't understand your question. The tree view uses them, no? And if so, use the ones used by the tree view for the Wiki.

luzpaz, 2022-08-27:
@wwmayer made a commit in a93997df64 but @PLChris is reporting that there is [still an issue](https://forum.freecadweb.org/viewtopic.php?p=591214#p591214)

luzpaz, 2022-08-27:
@ipatch can you test on macOS as well ? TIA!

ipatch, 2022-08-27:
> @ipatch can you test on macOS as well ? TIA!

@luzpaz you mind attaching a document that contains all the overlay icons in it, that way i can open the document and just screen cap what i see after opening the icons to make sure things are displaying as they should.

TIA!

luzpaz, 2022-08-28:
@PLChris ping Do you have an example pre-ready for ipatch ?

PLChris, 2023-02-21:
> @PLChris ping Do you have an example pre-ready for ipatch ?

Sorry for the delay, better ping chrisb on the forum.

Here is an example with a Cube and a Sketch as base objects. Cube001 and Sketch001 are links.
[LinkOverlay.zip](https://github.com/FreeCAD/FreeCAD/files/10789545/LinkOverlay.zip)

luzpaz, 2023-02-22:
@ipatch :point_up_2:

luzpaz, 2023-10-31:
@MisterMakerNL any chance you could tackle this issue ?

maxwxyz, 2024-03-03:
Any news on this issue?

maxwxyz, 2024-03-23:
it works fine on my side. Is this only a macOS error?
https://github.com/FreeCAD/FreeCAD/blob/2ebbd83268ffabf41ceaad802714a4ffd9214437/src/Gui/ViewProviderLink.cpp#L1708
The overlay icons seem kind of pixelated on Windows.
@wwmayer you've replaced XPM icons as .svg recently, do you know a fix?

maxwxyz, 2024-03-25:
it seems only a problem on macOS

JohnOCFII, 2024-10-12:
> it seems only a problem on macOS

I think you were asking me to test this on macOS.  I made a Standard Link from `TestBody` to `TestBody001` and I do see the icon for `TestBody001` is visably different than that of `TestBody.` **But** if each element in the Link (such as the sketch, pad, and pocket) icons in the `TestBody` are _also_ supposed to be visibly different than on the original -- I'm not seeing any difference.

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.50
```

[image]

PR #22359 (2025-07-06): Gui: Fix too small link icon on high dpi
Like #22345 it fixes issue with too small arrow on high DPI displays. The problem was that the code in ViewProviderLink used width that is already multiplied by DPR as size for icon that expects not premultiplied size, which was messing with proportions.

@maxwxyz can you test that it works as it should?

## Issues
Closes #22345
Fixes #6765

## Before and After Images

| Before | After |
|--------|--------|
| [image] |[image]  | 

Not sure why I get pixelated icons in that case, but I had
closed 2025-07-06 by hyarion