FreeCAD/FreeCAD #6766: [Bug] Navigation Cube -> Isometric View does not work

howie-j, 2022-04-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=68029

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Fedora Linux 35 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.28717 (Git) AppImage
Build type: Release
Branch: (HEAD detached at fee4046)
Hash: fee404602bee4e1d576cb6f979a8623c63c6815a
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.3.40
  * A2plus 0.4.56
  * Render 2022.1.0
  * FreeCAD_Assembly4 0.11.10
  * kicadStepUpMod 10.11.9
[/code]
```

### Issue description

When choosing Isometric in the dropdown menu under Navigation Cube, nothing happens. The other options (Ortho, Perspective, Zoom to fit) all work as expected. The toolbar icon also works.

[Similar issue reported in Link Branch, but seems to be fixed.](https://github.com/realthunder/FreeCAD_assembly3/issues/885)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-04-18:
This is a regression to 0.19.

luzpaz, 2022-04-18:
JFYI, there is another open Navicube related bug #6689

closed 2022-04-19 by wwmayer