FreeCAD/FreeCAD #6767: [Bug] Draft big Scale Multiplier with 2d text causes crash

m1cha1, 2022-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&p=589558#p589558

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.28717 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

Draft

### Issue description

In the Draft Workbench:
1. Create Text (for example "HELLO")
2. Change properties Display mode: 2D text
3. Change Scale Multiplier to 10000- you should get: Access violation
4. Increase Scale Multiplier to 1000000 or more you should get crash

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-19:
@donovaly I've added this to the 0.20 milestone because it's a crash

yorikvanhavre, 2022-05-11:
I can't reproduce the problem.. I can go to one billion without crash (Debian).
Maybe specific to windows?

donovaly, 2022-05-11:
I cannot reproduce either. I requested a recipe to reproduce in the forum.

m1cha1, 2022-05-16:
I still can reproduce in the latest version 0.20.28872

Below updated procedure:
1.Create Text (for example "HELLO")
2.Change properties Display mode: 2D text
3.Change Scale Multiplier by adding 0 after 1 without pressing Enter (10,10,100...) till crash 

In the log I have:
[code]
Log: Draft workbench activated.
Msg: ----------------
Msg: GuiCommand: Text
Msg: Pick location point
Log: make_text
Msg: ----------------
Msg: Text
Msg: string: ['HELLO']
Msg: placement: Placement [Pos=(-0.286788,0.496732,0.819152), Yaw-Pitch-Roll=(29.9999,-1.66922e-05,55)]
Err: Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Log: The event type 12 was sent to QOpenGLWidget
Object tree:
	QOpenGLWidget is child of
	SIM::Coin3D::Quarter::QuarterWidget is child of
	QStackedWidget is child of
	Gui::View3DInventor is child of
	QMdiSubWindow is child of
	QWidget is child of
	QMdiArea is child of
	Gui::MainWindow
[/code]

luzpaz, 2022-05-16:
@m1cha1 please give a reason why ticket was closed

m1cha1, 2022-05-16:
> @m1cha1 please give a reason why ticket was closed
I'm sorry my mistake

donovaly, 2022-05-17:
I stall cannot reproduce the crash with
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28886 (Git)
Build type: Release
Branch: master
Hash: a564d5ced7aecf320a0ac2d73f824a3dc7861eb3
Python 3.8.10, Qt 5.15.3, Coin 4.0.1, OCC 7.5.3
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
[/code]

maybe you file has an issue, therefore I requested a testcase. Can you please attach a simple FreeCAD file to a post in the forum?

m1cha1, 2022-05-17:
Test file added on forum (first post)

m1cha1, 2022-05-17:
I've loaded freecad to Windbg and recreate crash and I have a lot of this kind information:

[code]
3824.be4): C++ EH exception - code e06d7363 (first chance)
(3824.be4): C++ EH exception - code e06d7363 (first chance)
(3824.be4): Access violation - code c0000005 (first chance)
First chance exceptions are reported before any exception handling.
This exception may be expected and handled.
ig9icd64+0x73f50:
00007ffe`7ec53f50 f30f6f02        movdqu  xmm0,xmmword ptr [rdx] ds:00007fff`4c0bbab0=????????????????????????????????
[/code]

ig9icd64.dll is part of Intel HD Graphics Drivers for Windows so maybe this problem is related to my graphic card

yorikvanhavre, 2022-05-17:
There is most probably an issue somewhere related to how much space numbers are allowed to occupy in memory, which can differ depending on the OS, hardware, drivers... The fact that nobody else can reproduce the bug indicates it must be a rather specific combination...

I don't think there is anything we would be able to do on the FreeCAD side at the moment, but maybe let's keep this open in case new info arises?

luzpaz, 2022-05-25:
Discussion ongoing in the [forum thread](https://forum.freecadweb.org/viewtopic.php?f=3&p=589558#p589558)

luzpaz, 2022-07-09:
@m1cha1 please respond to https://forum.freecadweb.org/viewtopic.php?p=595873#p595873 (but instead of testing on the beta, please try testing on the latest stable v0.20 or even better v0.21 dev

Roy-043, 2022-07-10:
I can only confirm the issue.

When I increase the multiplier from 1 to 10 the text actually becomes smaller strangely enough.

The problem starts when the multiplier reaches 100,000.
There is no crash, but the text now looks strange. The spacing of the letters is off (In some cases only the "H" of "Hello" is visible or the letters appear superimposed), the text is much smaller than when the multiplier was 10,000 and It seems to have a serif font (instead of Arial).

Any change to the multiplier, even going back down to 10,0000, then causes FreeCAD to crash.

[image]

```
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.21.29484 (Git)
Build type: Release
Branch: master
Hash: 4de0e6fb198a60bd09400d64994e7aa8308b1e23
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods: 
  * BIM 2021.12.0
  * CubeMenu
  * Curves 0.4.4
  * fasteners 0.3.44
  * Help 1.0.3
  * Importer3D
  * InventorLoader 1.3.0
  * parts_library
  * ProDarkThemePreferencePack 1.0.0
  * TemplatePyMod
```

Roy-043, 2022-07-10:
... redoing the test with the version listed below I do not get a crash. Note the different Coin version.

There is a change in the font (to serif) and the text gets smaller when the multiplier reaches 100,000. But no letter spacing issue.

```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: Dutch/Netherlands (nl_NL)
```

m1cha1, 2022-09-07:
Crash with the latest weekly build with Coin 4.0.0
Could you please change the Coin version in the weekly build?

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30265 (Git)
Build type: Release
Branch: master
Hash: c35244e74f326a04c5bcab31e984742db582ac78
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: Polish/Poland (pl_PL)
[/code]

luzpaz, 2022-09-07:
>  Could you please change the Coin version in the weekly build?  

Please open ticket in https://github.com/FreeCAD/FreeCAD-Bundle

luzpaz, 2023-01-10:
@looooo is the weekly build still using coin 4.0.0 ?

looooo, 2023-01-10:
@luzpaz coin 4.0.0 is the latest version.

luzpaz, 2023-09-24:
According to @Roy-043 it's not coin3d related. So we need a diagnosis of what is happening here.

m1cha1, 2023-09-25:
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34387 (Git)
Build type: Release
Branch: master
Hash: d95b26dfb5dd23eaee004b4af3cf5cf4766e7f9c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
[/code]
On the latest version I still have an Access Violation and crash.

Roy-043, 2023-09-25:
> According to @Roy-043 it's not coin3d related. So we need a diagnosis of what is happening here.

You have misunderstood. The crash occurs on FreeCAD with Coin 4.0.0 but not on FreeCAD with Coin 4.0.1. At least that is my current diagnosis.

luzpaz, 2023-09-25:
Thanks for the correction. So upstream hasn't officially released 4.0.1 (https://github.com/coin3d/coin/issues/489)

luzpaz, 2023-11-18:
bumping upstream

m1cha1, 2023-12-08:
With the latest dev version no crash. Thanky you very much.
However, if you are in Display mode: Screen  and Scale multiplier over 8000 it won't increase the size of the text.
````
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35318 (Git)
Build type: Release
Branch: main
Hash: 855aa453a0122f30d03f6a0b7c88677312819b4c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```
`

closed 2024-05-14 by m1cha1