FreeCAD/FreeCAD #6772: [Feature Request] Support Height-map milling from image data.

sliptonic, 2022-04-20:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

None

### Subproject(s) affected?

Path

### Idea description

Many CAM applications can take a grey scale image and interpolate the color to a height map.  This is useful for producing lithophanes.  LinuxCNC has a image to gcode capability.  It would be nice if FreeCAD could generate this information from an image plane attached to a base object.

http://linuxcnc.org/docs/2.5/html/gui/image-to-gcode.html

### Anything else?

Came from a reddit discussion here.
https://www.reddit.com/r/FreeCAD/comments/u7g2n8/anyone_have_any_luck_using_the_cam_features_of/

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

boothdcb, 2022-11-08:
I too have been trying to do this, without success. I can create a solid from a lithophane, using the Part workbench, but then I can't integrate it into a model for milling. I'm interested in creating plaques for house signs, which incorporate the name of the house together with a carved image.

phaseloop, 2025-03-12:
There is a new tool to do that - seems cool:
https://github.com/wytr/OpenCarvePython

Core path generation logic is only 30 lines of Python, rest is UI and simulation

sliptonic, 2025-08-05:
What should be the desired workflow?

I can think of two possibilities:

1) The CAM workbench still works on models/solids.  The operation is aware of the image and can generate a tool path.
The user creates the model, imports an image, and attaches the image to a face of the model.  The the user selects the face and crates a  'image pocket' operation.  

This would be nice because the user could move the model around in the stock and the image (and tool path) would automatically update.

2) Modify CAM assumptions to allow the user to create a job with an image plane as the base object.  The user would still have to set job settings, stock, etc.   The operation would work directly on the image plane.

I prefer option #1 because I think it would be easier to extend in the future to allow the user to attach an image to a curved face.  It requires more upfront work from the user but reflects reality a bit better.

The fundamentals of the height map calculation are quite easy.  I've written a macro to do it.

Import an image.  adjust the macro to reference your image plane.

```
from PIL import Image
import numpy as np
from FreeCAD import Vector
import Mesh
from collections import defaultdict
from math import floor

def simulate_surface_from_points(points, res_x, res_y):
    """
    Creates a triangle mesh from a grid of 3D points.
    Assumes points are in row-major order, (res_x + 1) x (res_y + 1) grid.
    Flips rows to account for image coordinate system.
    
    Args:
        points (list of tuple): Flattened grid of (x, y, z) points.
        res_x (int): Number of segments in X.
        res_y (int): Number of segments in Y.

    Returns:
        Mesh.Mesh: FreeCAD mesh object.
    """
    mesh = Mesh.Mesh()
    idx = lambda i, j: j * (res_x + 1) + i

    for j in range(res_y):
        for i in range(res_x):
            # Flip Y axis by accessing rows from bottom up
            jj = res_y - j - 1

            p1 = Vector(*points[idx(i, jj)])
            p2 = Vector(*points[idx(i + 1, jj)])
            p3 = Vector(*points[idx(i + 1, jj + 1)])
            p4 = Vector(*points[idx(i, jj + 1)])

            mesh.addFacet(p1, p2, p3)
            mesh.addFacet(p1, p3, p4)

    return mesh

def imageplane_to_heightmap(image_obj, depth=5.0, invert=False):
    """
    Converts a FreeCAD ImagePlane object into a grayscale heightmap.

    Args:
        image_obj: FreeCAD ImagePlane object.
        depth (float): Total Z depth range from white to black (default 5.0mm).
                       Final Z will range from 0 to -depth (or +depth if inverted).
        invert (bool): If True, white pixels become deepest (Z = -depth).
                       If False (default), black pixels become deepest.

    Returns:
        dict: Heightmap data with:
            - z_map: 2D numpy array of depths (same size as image)
            - width_px, height_px: pixel dimensions
            - width_mm, height_mm: real-world sizes
            - placement: original FreeCAD placement
    """
    path = image_obj.ImageFile
    width_mm = image_obj.XSize
    height_mm = image_obj.YSize
    
    depth = abs(depth)

    img = Image.open(path).convert("L")
    arr = np.asarray(img).astype(np.float32)
    arr = (arr - arr.min()) / (arr.max() - arr.min())  # normalize to 0–1

    if not invert:
        arr = 1.0 - arr

    z_vals = -arr * depth  # Z = 0 to -depth

    return {
        "z_map": z_vals,
        "width_px": img.width,
        "height_px": img.height,
        "width_mm": width_mm,
        "height_mm": height_mm,
        "placement": image_obj.Placement,
    }

def generate_xy_grid(image_obj, res_x, res_y):
    width = image_obj.XSize
    height = image_obj.YSize
    placement = image_obj.Placement

    dx = width / res_x
    dy = height / res_y

    origin_x = -width / 2
    origin_y = -height / 2

    points = []
    for j in range(res_y + 1):
        for i in range(res_x + 1):
            local = Vector(origin_x + i * dx, origin_y + j * dy, 0)
            global_pt = placement.multVec(local)
            points.append((global_pt.x, global_pt.y, global_pt.z))

    return points

def map_heightmap_to_geometry(xy_points, heightmap):
    """
    Applies Z values from a heightmap to a grid of XY points.
    Automatically flips the image Y-axis and applies Z origin from the image placement.

    Args:
        xy_points (list of tuple): List of (x, y, z) points.
        heightmap (dict): Output of imageplane_to_heightmap.

    Returns:
        list of tuple: List of (x, y, z) points with corrected Z heights.
    """
    z_map = heightmap["z_map"]
    width_px = heightmap["width_px"]
    height_px = heightmap["height_px"]
    z_offset = heightmap["placement"].Base.z

    xs = [pt[0] for pt in xy_points]
    ys = [pt[1] for pt in xy_points]
    min_x, max_x = min(xs), max(xs)
    min_y, max_y = min(ys), max(ys)
    width_mm = max_x - min_x
    height_mm = max_y - min_y

    result = []
    for pt in xy_points:
        x, y = pt[0], pt[1]
        col = int((x - min_x) / width_mm * (width_px - 1))
        row = int((y - min_y) / height_mm * (height_px - 1))
        row = height_px - 1 - row  # Flip vertical
        row = max(0, min(row, height_px - 1))
        col = max(0, min(col, width_px - 1))
        z = z_map[row, col] + z_offset
        result.append((x, y, float(z)))
    return result

# --- Example usage ---

doc = App.ActiveDocument
iplane = doc.getObject("girl") # <== Name of your image plane

hm = imageplane_to_heightmap(iplane, depth=5.0, invert=False)
imageplane_to_heightmap(iplane)
grid = generate_xy_grid(iplane, 100, 100)
map = map_heightmap_to_geometry(grid, hm)

mesh = simulate_surface_from_points(map, res_x=100, res_y=100)
Mesh.show(mesh)
```

sliptonic, 2025-08-05:
> I see a "Map Reversed" checkbox, that's probably it. Great!

Yes I have one but that's not it