FreeCAD/FreeCAD #6775: [Bug] 3D Surface operation fails when Optimize Stepover Transitions is enabled in Multi-pass layer mode

ArifiKoseoglu, 2022-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=68145

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28717 (Git) AppImage
Build type: Release
Branch: (HEAD detached at fee4046)
Hash: fee404602bee4e1d576cb6f979a8623c63c6815a
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: English/United States (en_US)
[/code]
```

### Issue description

The sample model for this issue involves a pocket created by revolving a b-spline for 180 degrees and then cutting it out from a pad. (I am a newbie and in the forum discussion I learned that using this method was not the most efficient way of creating the model. However, the issue applies regardless.) The file is attached.

In using the "**3D Surface**" operation to carve out the volume, with "**Layer Mode**" set to "**Multi-pass**", enabling the "**Optimize Stepover Transitions**" option causes the path generation to fail with a "**list index out of range**" exception.

The report view window shows:

17:15:19  <Path.Area> Area.cpp(1436): hit bottom -0,0,1e-06
17:15:19  Traceback (most recent call last):
  File "/tmp/.mount_FreeCA68q2GR/usr/Mod/Path/PathScripts/PathUtils.py", line 71, in new_function
    res = function(*args, **kwargs)
  File "/tmp/.mount_FreeCA68q2GR/usr/Mod/Path/PathScripts/PathOp.py", line 819, in execute
    result = self.opExecute(obj)
  File "/tmp/.mount_FreeCA68q2GR/usr/Mod/Path/PathScripts/PathSurface.py", line 963, in opExecute
    self._processCutAreas(JOB, obj, idx, FACES[idx], VOIDS[idx])
  File "/tmp/.mount_FreeCA68q2GR/usr/Mod/Path/PathScripts/PathSurface.py", line 1075, in _processCutAreas
    final.extend(self._processPlanarOp(JOB, obj, mdlIdx, COMP, 0))
  File "/tmp/.mount_FreeCA68q2GR/usr/Mod/Path/PathScripts/PathSurface.py", line 1217, in _processPlanarOp
    self._planarDropCutMulti(JOB, obj, pdc, safePDC, depthparams, SCANDATA)
  File "/tmp/.mount_FreeCA68q2GR/usr/Mod/Path/PathScripts/PathSurface.py", line 1584, in _planarDropCutMulti
    nxtStart = ADJPRTS[i + 1][0]
<class 'IndexError'>: list index out of range
17:15:19  Recompute failed! Please check report view.

The issue does not happen when the "Optimize Stepover Transitions" option is **not** enabled.

[FCStd_GitHub.zip](https://github.com/FreeCAD/FreeCAD/files/8525901/FCStd_GitHub.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-04-24:
@sliptonic can you please have a look?

sliptonic, 2022-04-25:
I've asked @Russ4262 to have a look.

commit 5321edca (2022-05-05): Path: Fixes #6775 - Check for end of range

Change transition target point to last point at Safe Height if no more parts are available for processing.
commit b97e141e (2022-05-05): Merge pull request #6796 from Russ4262/fix/3d_surface_opt_step_trans

Path: Fixes #6775 - Check for end of range
closed 2022-05-05 by sliptonic