FreeCAD/FreeCAD #6776: [Bug] Compound transparency bug

DEL-TRONIK, 2022-04-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=589883#p589883

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux bookworm/sid (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.20.28717 (Git) AppImage
Build type: Release
Branch: (HEAD detached at fee4046)
Hash: fee404602bee4e1d576cb6f979a8623c63c6815a
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.11.9
[/code]
```

### Subproject(s) affected?

Part

### Issue description

The compound is not displayed correctly if we apply transparency :

[image]

If I delete the compound, everything is displayed correctly, So I think it's due to combine compound and transparency ?

Only silicon lens (red area) must get transparency.

Moreover if I export the compound with the transparency in .WRL format and that I add it to a component under KiCad, transparency it is not applied ..

### Anything else?

[compound_transparency_bug.zip](https://github.com/FreeCAD/FreeCAD/files/8529996/compound_transparency_bug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-04-21:
compound is not to blame, if you make a union instead of a compound its also buggy but less noticeable, presumably because there's less overlapping faces, according to yorik (https://forum.freecadweb.org/viewtopic.php?f=23&t=21144&start=10#p172110):

> it happens when a same shape has transparent and not transparent faces...

yorikvanhavre, 2022-05-11:
This bug happens when an object has a DiffuseColor property that contains color tuples with non-zero alpha values. It's either a bug in Coin3d, or something in it we are not using correctly. So far I've been unable to figure out what, though

DEL-TRONIK, 2022-05-15:
I just noticed a few things, I don't know if it may be useful, But when I export the previous file compound bug 3D model in .STEP format and then I open it by unchecking the option "Enable STEP Compound merge" in preferences, the export model is displayed correctly with the compound and transparency :

[image]

[image]

If the "Enable STEP Compound merge" option is checked, it does not work, and the result is as before (the bug compound and transparency).

closed 2024-12-29 by github-actions[bot]