FreeCAD/FreeCAD #6793: Snapping feature will not work under certain conditions

xStreeme, 2022-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=67463

### Version

0.19 (Release)

### Full version info

```shell
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git)
Build type: Release
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: C/Default (C)
```

### Issue description

I cannot snap.
Neither the snap icon nor the white dot will appear.
It looks like I encounter this issue while working with my Mac, not when working on my wife's Win11 laptop.

I really have no clue of where the problem is originated, however, as I tried to explain in my topic, **it happens whenever the system cursor is shown instead of the crosshair cursor**.
As I described in my posts, it _may_ not happen if I turn on the computer and don't connect the external display (via USB-C). However, with the same display connected to the Win11 laptop (via HDMI instead), I will **not** experience such issue.

It looks like no-one on the forum could help.
I may use the Win11 computer as a workaround, hoping in a fix.
Unfortunately I can't code, but I may give developers further informations about this problem, if needed.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-25:
Why was this ticket created without testing in v0.20dev ?

xStreeme, 2022-04-25:
Because it behaves exactly the same way.

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.28751 (Git)
Build type: Release
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: C/Default (C)
[/code]

All existent tickets seem oriented to issues that have not much to do with the one I'm reporting.

[image]

tijtij, 2022-11-06:
I was able to get the crosshairs cursor to appear and get snapping to work normally by going into macOS' System Preference and selecting **Default for Display** as the display resolution. Neither changing the resolution back to **Scaled**, rebooting macOS, or attaching an external display caused the bug to reappear.

The only issue I am still experiencing is that snapping only works on one side of an object, unless of course snapping is intended to behave like that and therefor not an issue.

[image]

kamazeuci, 2023-01-30:
I think this is the same issue I am having, but on GNU/Linux with wayland. Could this be a QT/Kvantum config issue?
Please see here my post and the link to the video:
https://forum.freecadweb.org/viewtopic.php?f=8&t=75678

xStreeme, 2023-04-06:
Hello! Any news on this issue?
I installed the current version but the crosshair cursor still does not appear instead of the system cursor.

seankanderson, 2023-07-16:
I ran into this as well. Can't trigger the work-around by switching resolutions. Stuck working on my Windows Desktop.  :(

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * BIM 2021.12.0
[/code]

luzpaz, 2023-08-16:
Needs test on latest stable v0.21

Thenicolaibulow, 2023-09-14:
Confirmed to still be an issue on v0.21.1-33694

I'm on a M1 Pro 16", and seemingly no snapping tools work.
No luck with the display scaling workaround. 

Back to proprietary CAD is it..

Roy-043, 2023-09-16:
Related forum topics:
https://forum.freecad.org/viewtopic.php?t=67997
https://forum.freecad.org/viewtopic.php?t=81009

Roy-043, 2023-09-17:
I have a possible solution that I would like to see tested:

Edit:
`<Install_folder>/Mod/Draft/draftguitools/gui_snapper.py`

Remove these two lines:
```
        if int(QtCore.qVersion().split('.')[0]) > 4:
            new_icon.setDevicePixelRatio(device_pixel_ratio)
```

Restart FreeCAD and check if the issue with the offset cursor persists.

Thenicolaibulow, 2023-09-21:
@Roy-043 - thanks for the suggestion. Seems like these lines did, definitely influence (and improve) the snapping abilities. 

I am now doubting how prevalent my original issue was. It seems now, that don't enable more than a couple of snapping tools at a time; the snapping works alright. 

I have a hard time gauging how well it works, as I am used to Rhino3D on a daily basis - which has fantastic snapping abilities. I might be biased!

PR #10962 (2023-10-09): Draft: Fix cursor scaling issue
Fixes #6793
commit 80f97a58 (2023-10-09): Draft: Fix cursor scaling issue

Fixes #6793
closed 2023-10-09 by yorikvanhavre