FreeCAD/FreeCAD #6799: [Bug] Can't exit Stereo view

luzpaz, 2022-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=68314

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.20.28753 (Git)
Build type: Release
Branch: master
Hash: fe5ac277a017413a8646b621d5b993b0c606d551
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: Bulgarian/Bulgaria (bg_BG)
```

```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28756 (Git)
Build type: Release
Branch: master
Hash: 53cc22225599a07addfbdc7c9b560795dbd80a83
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.6.1
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

Core

### Issue description

### Steps to Reproduce:
* Create new documemt
* View > Stereo > (Choose any stereo mode)
* View > Stereo > `Stereo off`

Result:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'NameError'>: Unknown stereo type 'None
```

### Anything else?

First reported by @kisolre

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-04-27:
That is as far as I got: https://github.com/FreeCAD/FreeCAD/blob/cb1093ef305010d218c74868e65d64e261fce3fb/src/Gui/CommandView.cpp#L2285

chennes, 2022-04-27:
The problem is that "None" is not actually the value in that enumeration: it's "Mono".

closed 2022-04-27 by chennes

luzpaz, 2022-04-27:
@chennes just for my edification, where is the enumeration located in the code?

commit f8f3ba7a (2022-04-27): Gui: Fix Stereo view disabling

Fixes #6799.