FreeCAD/FreeCAD #6803: [Core] Base/FileInfo.cpp: To avoid race conditions we should rather return a file pointer than a file name

luzpaz, 2022-04-27:
https://github.com/FreeCAD/FreeCAD/blob/cb1093ef305010d218c74868e65d64e261fce3fb/src/Base/FileInfo.cpp#L138-L141

nebularnoise, 2022-10-18:
Hi! I'm interested in taking this one (trying hacktoberfest for the first time)

In almost all case, it seems this function is called to instantiate an fstream, so it would seem to make sense to return an opened fstream directly. In a few other cases however, it seems only the name is needed, and the file should remain closed.

There could either be two functions, one returning the name, to maintain 'legacy' behaviour, and the other returning a fstream.
Alternatively, std::fstream could be wrapped to also store the filename.

luzpaz, 2022-10-18:
@wwmayer could you weigh-in on this?

luzpaz, 2022-10-26:
Hey @berniev could you take a stab at this?

berniev, 2022-10-26:
@luzpaz Thank you for your thinking of me, but to be honest this question is still outside the realm of my knowledge and experience.

luzpaz, 2022-11-16:
@nebularnoise this maybe too ambitious. We need @wwmayer to weigh-in. But in the meantime, feel free to choose another issue?

wwmayer, 2022-11-18:
The function getTempFileName() is supposed to return the name of a tmp. file. Therefore it uses some OS-specific functions that already create a handle of a tmp. and release it immediately (Windows: GetTempFileName / DeleteFile and other OS mkstemp / unlink) just to get this name.

The problem is that when trying to open the file for writing there is no guarantee that in the meantime any other process already has access on this file. The consequence may be that the access will be denied or even worse the file will be (partially) overridden.

The better way would be to return a class object (e.g. TemporaryFile) where its constructor creates the file and the destructor releases it.

CalebHeydon, 2024-04-10:
I was looking through the codebase to see where this function is used, and it looks like there are a few places where something is appended to the temporary filename. I was wondering what the best way to handle that would be? Here's an example: https://github.com/FreeCAD/FreeCAD/blob/cc96f2718e139d670e08c4226bdea9955a102ed6/tests/src/Mod/Mesh/App/Exporter.cpp#L22

nebularnoise, 2024-04-11:
@CalebHeydon 
In terms of the exposed API, maybe adding a "suffix" argument would work? 
On top of that, as @wwmayer suggested, if the function returns a TemporaryFile object, it would prevent future "abuses" of this function call which modify the filename.

In terms of implementation, `mkstemps` supports suffixes out of the box. For Windows, I'm not aware of a system call which does, but I'm no expert. In any case, However, if we're to return a file handle, it will be necessary to check whether the file creation is successful, so we might as well move this "append" within the implementation, and try to create the file until successful.

CalebHeydon, 2024-04-11:
Adding a suffix argument is a good idea. What would be the benefit of creating a TemporaryFile wrapper over returning an fstream? File deletion?

nebularnoise, 2024-04-12:
Yes, I assume in many cases, the file is ultimately deleted. For that, an fstream is inadequate.

The original comment suggests one workaround, using a `std::FILE*` directly. 
https://github.com/FreeCAD/FreeCAD/blob/cb1093ef305010d218c74868e65d64e261fce3fb/src/Base/FileInfo.cpp#L140-L141
While that does solve both problems of being able to delete the file, and to maintain ownership of it to prevent race conditions, it leaves the function very vulnerable to misuses, because returning a `std::FILE*` gives too much power to the caller. 
There's no way to prevent them from copying the filename and closing the file immediately. From their perspective, it would be a very convenient thing to do, it's the same behaviour as the old function, they got a string containing the name of a temporary file. But we'd be back to square one.

Wrapping a `std::FILE*` into a `TemporaryFile` would provide a much friendlier interface, with `unique_ptr`-like "lifetime management", and prevent reintroduction of similar race condition issues.

CalebHeydon, 2024-04-12:
I not a C++ expert. How would you implement the TemporaryFile lifetime management? Why not just have the function return a unique pointer?

nebularnoise, 2024-04-13:
`unique_ptr` applies the RAII philosophy to memory allocation, this class would do the same but for files on the filesystem.

I've been playing with the idea on Compiler Explorer, here's what I've got so far
https://godbolt.org/z/9cGxWoW98

maxwxyz, 2024-07-01:
@chennes could you have a look on this?