FreeCAD/FreeCAD #6809: [Feature Request] CLI -t flag should list all available unit tests (instread of user guessing which to run (besides `-t 0` AKA all tests)

luzpaz, 2022-04-28:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

n/a

### Subproject(s) affected?

Core

### Idea description

When running FreeCAD executable from the CLI with the `-t` flag, user should get a list of optional individual tests to choose from instead of needing to guess or lookup the names for them.  
[image]

Current workaround:  
I need to start FC GUi and go to the TestWB and find the name of the individual test that I want to run (not very convenient)
[image]
 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-05-15:
Here's the GUI code used in TestWB to get the tests:  
https://github.com/FreeCAD/FreeCAD/blob/aa19e216625cbfdab5099d8eea6b62122917a3f6/src/Mod/Test/Gui/UnitTestImp.cpp#L254-L260

CC @wwmayer

luzpaz, 2022-05-16:
@mwganson i was wondering if you can help me figure out how the UnitTestImp.cpp is getting the names of all the tests. It looks like the names are found in the `src/Mod/$foo/init.py` files, for instance the AddonManager test files 
(https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/AddonManager/Init.py#L8)
 
`FreeCAD.__unit_test__ += ["TestAddonManagerApp"]`

mwganson, 2022-05-17:
Sorry, I'm not feeling well right now.

ageeye, 2022-05-18:
You can clone [TestApp.All](https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Test/TestApp.py#L55):

```
def PrintAll():
    # Registered tests
    tests = FreeCAD.__unit_test__

    print("Registered test units:")

    for test in tests:
        print(test)
```

Untested:
```
if (testCase.empty()) {
    testCase = "TestApp.PrintAll";
}
```

yorikvanhavre, 2022-05-18:
From the command line, though, not all tests are present I think:
```
>>> FreeCAD.__unit_test__
['MeshTestsApp', 'TestFemApp', 'TestDraft', 'BaseTests', 'UnitTests', 'Document', 'Metadata', 'UnicodeTests', 
'TestPythonSyntax', 'TestSketcherApp', 'TestAddonManagerApp', 'TestSpreadsheet', 'TestTechDrawApp', 
'TestPathApp', 'TestPartApp', 'TestOpenSCADApp', 'TestPartDesignApp']
```
Of course a quick fix would be to hard-code the list in the help text shown to the user...

ageeye, 2022-05-18:
> From the command line, though, not all tests are present I think:
>
> ```
> ['MeshTestsApp', 'TestFemApp', 'TestDraft', 'BaseTests', 'UnitTests', 'Document', 'Metadata', 'UnicodeTests', [...]

Yes. "./FreeCADCmd -t" only display which do not need the GUI. But "./FreeCAD -t" lists all test units. 

[First Draft](https://github.com/ageeye/FreeCAD/commit/698a585740ff84074fcb1cf47b54a990576f85b3)

luzpaz, 2022-05-24:
Nice @ageeye  
Would you be open to making a PR? :grinning: 

@mwganson hey mate, hope you're feeling better. I was AFK for a while myself so I didn't see your response before.

ageeye, 2022-05-25:
> Nice @ageeye Would you be open to making a PR? 😀

Done

PR #6915 (2022-05-31): App: CLI -t flag list all available unit tests
Without parameter all available test units will listed. I hope the hints clarify the using.

---

Closes #6809
closed 2022-05-31 by donovaly