FreeCAD/FreeCAD #6810: FreeCAD fails to open OpenSCAD files.

felixknollmann, 2022-04-28:
When opening a OpenSCAD file I get the following error:
16:27:03  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.19\Mod\OpenSCAD\importCSG.py", line 119, in open
    processcsg(tmpfile)
  File "C:\Program Files\FreeCAD 0.19\Mod\OpenSCAD\importCSG.py", line 177, in processcsg
    result = parser.parse(f.read())
  File "C:\Program Files\FreeCAD 0.19\bin\lib\site-packages\ply\yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "C:\Program Files\FreeCAD 0.19\bin\lib\site-packages\ply\yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "C:\Program Files\FreeCAD 0.19\Mod\OpenSCAD\importCSG.py", line 854, in p_multmatrix_action
    parentcolor=p[6][0].ViewObject.ShapeColor
<class 'IndexError'>: list index out of range

Instead of opening the part with all subsections, it just makes a single cube.

felixknollmann, 2022-04-28:
I tested with the 0.20 branch and had the same result. Also used 0.19.1 (originally using 0.19.4)

felixknollmann, 2022-04-28:
Here is an example (scad files uploaded as .txt) The first is a dependency of the second.
[thorlabs_optomech.txt](https://github.com/FreeCAD/FreeCAD/files/8587232/thorlabs_optomech.txt)
[baseplate_test1.txt](https://github.com/FreeCAD/FreeCAD/files/8587233/baseplate_test1.txt)
:

luzpaz, 2022-04-29:
Next time, just zip scad files in to 1 archive + please add your full FreeCAD [About](https://wiki.freecad.org/About) info when reporting issues.

luzpaz, 2022-04-29:
Honestly, I think we should close this ticket and post the issue to the [dedicated forum thread for OpenSCAD issues](https://forum.freecadweb.org/viewtopic.php?f=3&t=61777)

felixknollmann, 2022-04-29:
Here is the full FreeCAD about info for my version: 
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git)
Build type: Release
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: C/Default (C)

I also tried versions 0.20 dev, 0.19.4, and 0.19.1 on Windows 10.

luzpaz, 2022-04-29:
Closing in favor of https://forum.freecadweb.org/viewtopic.php?f=3&t=61777&p=592113#p592113

closed 2022-04-29 by luzpaz

luzpaz, 2022-05-16:
@felixknollmann please retest on the latest v0.20dev 
Issue was not reproducible with: 
```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28764 (Git)
Build type: Debug
Branch: master
Hash: 4f9eb9ebcea67bff2a46cfc97ae125472f115a11
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.3
Locale: English/United States (en_US)
```