FreeCAD/FreeCAD #6815: [Bug] Path: 3D Pocket does not develop path to bottom when selecting multiple surfaces

olddude56, 2022-04-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=68055&sid=286d3ed5ac4994e282405891d8605980

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

### Issue description

When using 3D pocket to tool path  a pocket with multiple surfaces and depths, the toolpath will be developed to the "shallowest" depth.  When selecting the deep surfaces only, the toolpath will go to the bottom.  Images, models and examples in forum discussion for reference.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Russ4262, 2022-05-08:
As a participant in the forum thread linked in the issue description, I would like to add that this error reported appears to be limited to models created in Part Design workbench - at least this is my experience so far.

maxwxyz, 2024-12-21:
@sliptonic @olddude56 is this issue still relevant?

closed 2025-06-08 by github-actions[bot]

tcurdt, 2025-06-08:
This is still relevant - isn't it?

luzpaz, 2025-06-11:
@olddude56 sorry, your response is ambivalent to me. Is the issue still reproducible ?