FreeCAD/FreeCAD #6818: [Feature Request] Add an option to lock the NaviCube

wwmayer, 2022-05-02:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=34&t=68390

### Idea description

Sometimes it happens that the NaviCube is dragged by accident. To avoid this an option to lock it would be nice.

In the preferences dialog a check box can be added that affects all existing and newly created NaviCubes. Then in the context-menu there could be this option too but only affects this instance.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jolbas, 2023-03-31:
@wwmayer  I can take a look at this. But I don't understand the need to set the lock individually. Should that be persistent by file then? I would rather change the behaviour of the current corner options to globally lock the navigation cube to a corner and then have a fifth 'drag to place' or 'draggable' option in popup.

wwmayer, 2023-03-31:
> But I don't understand the need to set the lock individually.

It can happen that a user has set the global option to lock navi cubes but exceptionally wants to tmp. unlock one instance. Then it's very inconvenient having to go to the preferences, make the change and go back to the preferences.

> Should that be persistent by file then?

No. The global option is defined in a parameter group and the NaviCube class has a boolean member that is read from the parameter group at construction time. There are setter and getter methods to change the boolean member variable.

> I would rather change the behaviour of the current corner options to globally lock the navigation cube to a corner and then have a fifth 'drag to place' or 'draggable' option in popup.

Not sure if I got you right: so you want that NaviCube by default is locked and in the context-menu you can tmp. unlock it. Is that right? If yes, sure that's also an option.

Jolbas, 2023-03-31:
> Not sure if I got you right: so you want that NaviCube by default is locked and in the context-menu you can tmp. unlock it. Is that right? If yes, sure that's also an option.

Not exactly what I ment but that is better. I try to implement this and maybe I discover something more on the way.

Jolbas, 2023-04-06:
@wwmayer I need some advice. There are two observers of the navicube preferences. The NaviCubeSettings class in Gui/View3DSettings and the NaviCubeImplementation class in Gui/NaviCube. Can I remove one of the observers or are there reasons to have both. They seem to do the same thing and every parameter is set twice.
https://github.com/FreeCAD/FreeCAD/blob/8ea5d9bac10698a9ae31735ad45a63688a6be58b/src/Gui/NaviCube.cpp#L391
https://github.com/FreeCAD/FreeCAD/blob/8ea5d9bac10698a9ae31735ad45a63688a6be58b/src/Gui/View3DSettings.cpp#L454

wwmayer, 2023-04-06:
> Can I remove one of the observers or are there reasons to have both.

I vote to remove NaviCubeImplementation::OnChange and the inheritance from ParameterGrp::ObserverType. If NaviCubeImplementation::OnChange handles some parameters that are not handled by NaviCubeSettings then they should be moved.

NaviCube(Implementation) shouldn't have any direct access to a parameter group any more and instead provide setter functions. This will make the code much clearer and reduce side-effects.

Jolbas, 2023-04-06:
Ok. Thanks.

Jolbas, 2023-04-10:
I added an item to the NaviCube menu called "Move NaviCube". If selected you have a single opportunity to drag and move the cube. Then its locked again and you have to select the menu item before every move. Is this ok? Or should it be draggable until actively locked?
[image]

wwmayer, 2023-04-10:
> Or should it be draggable until actively locked?

I prefer this option because the other option might be confusing to people.

wwmayer, 2023-04-10:
> Move Navicube

Btw, in the menu item you should write "Move navigation cube" because in the UI we nowhere use Navicube or NaviCube but the  full word "navigation cube". (NaviCube is only used for the name of the parameter group or API functions.)

closed 2023-04-24 by chennes

Bryfour, 2023-05-27:
I think that if the cube was locked as a default and I could click and drag it if I used the little dot that's already there it would be perfectly fine.
and for people who want to just click and drag it as is, they can go into the settings...

PR #9356 (2023-04-24): Refactor Navigation Cube
This PR does not conform to the one issue recommendation. Hopefully you can bear reviewing it anyway.
Fixes #9262 
Fixes #6935 
Fixed #6818 
But also a non registered issue from the forum:
[https://forum.freecad.org/viewtopic.php?t=77768](url)
Will hopefully also fix a problem on Raspberry Pi
[https://forum.freecad.org/viewtopic.php?t=76438](url)

Here is a list of changes
- Uses std commands in menu coming with icons
- Cube only draggable if activated in menu 
- [image]
- Ortographic and Perspective reflected in Navigation Cube
- [image]
- [image]
- Remove textures ex [... cut]