FreeCAD/FreeCAD #6824: [Bug] Crash on startup

mdziczkowski, 2022-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

na

### Version

0.19 (Release)

### Full version info

```shell
Version: 0.19
Build: 24366
Downloaded from: FreeCAD website (as a AppImage)
```

### Issue description

While starting, FreeCAD has crashed with following error:

```
Program recieved signal SIGSEGV, Segmentation fault.
#0 /lib/x86_64-linux-gnu/libc.so.6(+0xcbd60) [0x7f722b9bbd60]
```
The initialization strings in the log file look normal, except the last lines containing above error message

### Anything else?

### Enviroment:

**OS:** Debian 11
**Python version:** 3.9
**GUI:** X11 + openbox
**OpenGL version:** 2.1 (2.1 Mesa 20.3.5)

### Other informations:

Because sending this report from a phone, the most important logs part has been rewroten here

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-05-02:
Closing in favor of https://github.com/FreeCAD/FreeCAD-Bundle/issues/102#issuecomment-1113631223

closed 2022-05-02 by luzpaz

mdziczkowski, 2022-05-03:
> _Edit: next time please open a forum thread instead of a ticket. thank you in advance_

Hello.

First, on forum is no seperate sub-category for neither bug reports, nor feature requests and for second, your request stands in opposite to a forum entry, where stands that on developrs may miss it there, so they write to open a issue here.

luzpaz, 2022-05-03:
Help subforum is for Bugs.  Open subforum is for FR. 

FC is complex due to it's many 3rd party dependencies. We prefer users not rely on their distro's default FC build or that they know what they're doing when building from source because since too many bugs are produced due to conflicts/mishaps.