FreeCAD/FreeCAD #6831: [Feature Request] Block constraint for vertices

FEA-eng, 2022-05-04:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=68434

### Subproject(s) affected?

Sketcher

### Idea description

Currently, the block constraint doesn't work for free points (reported here: #5844) but it would also be good to make it work for vertices (points belonging to other geometrical objects).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-03-28:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/5844 it seems

FEA-eng, 2024-03-28:
> Duplicate of #5844 it seems

Not really. I've mentioned #5844 in the description of this issue. The older one is about free points (Sketcher CreatePoint) while this one is about vertices of lines or other geometrical entities.

PaddleStroke, 2024-03-28:
Ah sorry I have not seen you mentioned it already.

AjinkyaDahale, 2024-05-10:
Isn't this (and #5844) just the lock constraint?

FEA-eng, 2024-05-11:
> Isn't this (and #5844) just the lock constraint?

Lock constraint only applies horizontal and vertical dimensions automatically (the same could be done manually). Block constraint doesn't apply any dimensional constraints and just makes the line immovable (helpful for debugging or quickly and temporarily constraining the sketch in general).

AjinkyaDahale, 2024-05-21:
> Lock constraint only applies horizontal and vertical dimensions automatically (the same could be done manually). Block constraint doesn't apply any dimensional constraints and just makes the line immovable (helpful for debugging or quickly and temporarily constraining the sketch in general).

But for vertices this has the same effect.

FEA-eng, 2024-05-21:
> But for vertices this has the same effect.

In both cases, a geometrical entity ends up fixed but Lock adds 2 editable dimension constraints to a vertex while Block is a single non-editable constraint applied to the whole line. If it was possible to apply it to vertices as well, there would be a quick way to fix multiple vertices in place, without adding any actual dimensions. Lock is basically just a mode of the new Dimension tool now - it automatically applies 2 dimension constraints linked to the origin without asking for values. Block is better for testing/debugging without cluttering the sketch with more dimensions.

SolidWorks also lets you fix a vertex in place without dimensioning it:

[image]