FreeCAD/FreeCAD #6835: Path: Make conditional debug code uniform throughout the Path WB code

luzpaz, 2022-05-05:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

gitter

### Subproject(s) affected?

Path

### Idea description

Per LGTM, many PathWB conditional DEBUG rules are "_unreachable_". 

"_Unreachable code makes the code more difficult to understand and may slow down loading of modules._"   

See https://lgtm.com/projects/g/FreeCAD/FreeCAD?mode=tree&ruleFocus=3980095  

When I spoke to @Russ4262 on gitter (i'm not sure if that is accurate) we understood that not all the DEBUG methodology is uniform in the Path code base. This ticket is therefore opened in order to track this issue so we can get it resolved eventually.  

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-05-12:
[image]

@Russ4262 do you have any interest working on this issue. I'd love to see the LGTM results diminished as best as possible as we work to improve uniformity in FC code.

Russ4262, 2022-05-13:
I browsed a few of the files, a large percentage use the `if False:` code block for the debugging disablement. Perhaps project supervisors have chosen it for readability and simplicity over LGTM concerns?

I might have some time to go through and standardize some or all of the affected Path files if we decide on a syntax replacement.

Thanks for all you do for FreeCAD and its community.

luzpaz, 2022-05-14:
@Russ4262 thanks for looking through this. I agree, if there is no necessity to do this...we don't have to appease LGTM. But it seems to me that having a uniform debugging approach for Path would make sense. 

CC @sliptonic (Care to weigh-in when you get a moment?)

sliptonic, 2026-09-22:
No action needed at this time.

closed 2026-09-22 by sliptonic