FreeCAD/FreeCAD #6836: [Bug] <title>

Buffalo1John, 2022-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?t=68353

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.28793 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: English/United States (en_US)
[/code]
```

### Issue description

the behaves the same way in 19.3 release and all of the versions of 0.20 that I have tried for the last few months. I have created a 40mm cube and rotated it on the screen. In the forum discussion, there are examples of the improper rendering.

I have seen this failure with the cube created in Part Design and also in Part. I have verified that the same failure mechanism happens for more complex designs, but the simple cube most easily reproduces the problem. If the part is saved when it displays the problem, opening it in any of the versions from 0.19.3 through the current  version show the same improper display.

I have changed the display preferences to use each and every render cache option, every anti-aliasing option, and transparent objects options. Additionally, I have tried checking "Use software OpenGL", restarted the program, and the problem persists.

[t1p.zip](https://github.com/FreeCAD/FreeCAD/files/8635983/t1p.zip)
[image]
.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Buffalo1John, 2022-05-05:
Forgot to add this, I can make the improper rendering go away, but it can come back it the object is rotated some more. 

I am using Orthographic rendering. If I switch to Perspective rendering, I can't get the display problem to reappear.

luzpaz, 2022-05-05:
Looks related to ["Corner appears cut"](https://forum.freecadweb.org/viewtopic.php?f=3&t=64833) thread.

Buffalo1John, 2022-05-05:
Similar - maybe...

The corner cut seems to be on just the one corner, but it does not change as the rotation changes. And if you rorate slightly, the cut goes away. 

Just tried the 40 mm cube and if I switch to the Isometric view, the corner is cut, and interestingly, you can see the inside corner because my cube is a cube.
[image]

The big difference is that the corner cut is a small plane cut and does not move. On my corners cut, the plane appears to me to be about the same angle, but the plane seems to move on an axis perpendicular to the plane, to the point that it can cut most and even all the object (the cube).

Buffalo1John, 2022-05-07:
More information:
I colored each face of the cube with a different color. It seems that it is possible to create this improper rendering with any cube corner. The cube is sliced by the clipping plane and parts on one side of the plane are visible while the parts on the other side of the clipping plane are hidden. The hidden parts are facing the screen but it is not clear if this clipping plane is attached to something else.

When you have the clipped cube, if you move the cursor over one of the faces, it is highlighted, but only when the visible part of the cube face. In the first image, the cursor is on the red dot, which is not on the visible part of the cube face. 
[image]

In this next image, the cursor is also on the red dot, but that part of the cube face is visible, so the face has been highlighted.
[image]

It was suggested that if you get it into the clipped state, if you click on the cube before beginning the rotation, the full cube reappears. However, if one continues rotating without clicking on the cube, the visibility cutting plane will come back.

Please note in all of this, saving the model in the clipped state, when reloaded will still have the clipped state. Perhaps ther is a tool which dumps the file contents which would reveal this clipping plane, but I don't know of such a tool...

adrianinsaval, 2022-05-07:
IMO it's the same issue, it always seemed random to me, I can't pinpoint the required conditions for it. My guess is that this is a bug in Coin3D rather than FreeCAD.

luzpaz, 2022-08-27:
@wwmayer was [debugging](https://forum.freecadweb.org/viewtopic.php?p=593878#p593878) coin bug

adrianinsaval, 2023-05-04:
@Jolbas I think you've been playing around lately with opengl rendering depths right, maybe you have some idea of what is causing this or how we could solve it?

Jolbas, 2023-05-04:
Not really. Sorry. If I could have reproduced it then I probably could have found a reason. And I'm not experienced enough to maka a good guess. But there have been some fiddling with keeping OpenGL state stable the last few days so it may be worth just try again. Can @Buffalo1John or someone else still make it happen?

wwmayer, 2023-05-04:
The issue is only about the clipping planes of the view frustum but I don't know how it happens. A way for the user to fix it is to run the "fit all" function.

Buffalo1John, 2023-05-04:
I'm running 0.21
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32985 (Git)
Build type: Release
Branch: master
Hash: 5cd377f931312ede305e4c764ba2028b20eaacfd
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods:
  * Assembly4 0.50.1
[/code]

I drew the same simple object, rectangle 50x47mm in a sketch, padded to
40mm. I can't make the disappearing corners happen with the above release.

So, I loaded one of the original test files I created 2022-04-28T17:06:52Z
and the 0.21 release (above) and the missing corner is there. I agree it is
likely a clipping plane, but it seems to me it is in the file.

[image: image.png]
So, then I tried the View/StandardViews/FitAll and the view was corrected
and I can't make the messed up view come back. A year ago, it was easy with
a few rotations of the object to make it mess up, now - I can't reproduce.

Jolbas, 2023-05-04:
If I open one of the example files the initial rendering also shows the error but it is immediately disappearing on any kind of screen update. It's as if the old rendering is stored in the file.

dominikh, 2023-05-10:
I can still reproduce it with both FreeCAD and Coin3D built from their respective master branches (ecb72b8eadced103cc648990f8ebe251d64c3f43 and 4c67945a58d2a6e5adb4d2332ab08007769130ef at the time of writing), on newly created files.

I do find it significantly easier to trigger the bug when moving erratically using a Spacemouse.

[image]

FlachyJoe, 2023-05-10:
Step to reproduce :
- Zoom out
- Pan the view to have the cube near a side of the screen
- Rotate the view from the other side

adrianinsaval, 2023-05-10:
That's it! Finally reproduction steps, looks like you need to rotate until the cube goes partially outside the view on the other side then when it comes back to view it's clipped.

adrianinsaval, 2023-05-10:
Looks like it's a problem on how we zoom out, we may be missing some command to update the visible range for the objects, the farther away you zoom the easier it is to reproduce

adrianinsaval, 2023-05-10:
potentially meaningful:
https://github.com/FreeCAD/FreeCAD/blob/ecb72b8eadced103cc648990f8ebe251d64c3f43/src/Gui/Quarter/QuarterWidget.cpp#L301

https://www.coin3d.org/Coin/html/classSoRenderManager.html#aa73c943c31d9d7950858fc2ef24ea45f
maybe we should be using FIXED_NEAR_PLANE ?

Buffalo1John, 2023-05-10:
I think you need to do both rotate and move. If you only rotate, nothing
goes clippy. If you rotate, then move, then rotate, it's clipping. I didn't
zoom at all and I verified with multiple navigation styles. (Gesture style,
only left and right mouse buttons).
[image: image.png]

Once it gets clippy, only rotate is needed to make the view point and
clipping plane move about.

Jolbas, 2023-05-11:
> maybe we should be using FIXED_NEAR_PLANE ?

That doesn't help
It seem to rather be the camera position and everything behind the camera is clipped.
Does this clipping make sense in orthographic view? If you want to look inside a building it is necessary. Then we need to decide how to control it and make a difference between move camera forward and zoom. Currently a zoom operation desn't seem to change camera position when in orthographic view.

https://github.com/FreeCAD/FreeCAD/blob/ecb72b8eadced103cc648990f8ebe251d64c3f43/src/Gui/NavigationStyle.cpp#L705-L708

Jolbas, 2023-05-11:
This could be fixed by inserting
```
    if (cam->getTypeId().isDerivedFrom(SoOrthographicCamera::getClassTypeId())) {
        cam->focalDistance = static_cast<SoOrthographicCamera*>(cam)->height;
    }
```
here
https://github.com/FreeCAD/FreeCAD/blob/ecb72b8eadced103cc648990f8ebe251d64c3f43/src/Gui/NavigationStyle.cpp#L625

But there are related problems linked to this.
- view to fit cube
- zoom out and pan to get the cube on the side
- rotate the view so that the cube is as far away as possible
- zoom in on the cube
- try to rotate the view
The focal point that the is the pivot point is a long distance in front of the cube.
The problem is less prominent if Preferences->Display->Navigation->Rotation Mode is set to anything else than "Window center"

wwmayer, 2023-05-11:
When you observe the clipping issue you can print the current camera settings with `print(Gui.ActiveDocument.ActiveView.getCamera())`. 

You will see that the value of the nearDistance is negative and farDistance has a small but positive value that is less then the value of the focalDistance. Normally all distances are positive with this relation:
`0 < nearDistance < focalDistance < farDistance`

There are two ways that helps to improve the issue:
* make the nearDistance even smaller (but I don't understand yet the logic behind)
* move the camera position in opposite view direction

luzpaz, 2023-05-30:
Anyone want to take a stab at implementing wwmayer's suggestions ?

adrianinsaval, 2023-06-01:
This very long standing issue is now finally solved! Thanks @Jolbas

closed 2023-06-01 by adrianinsaval

Buffalo1John, 2023-06-01:
Thanks @Jolbas <https://github.com/Jolbas> --- Thank you!!!

adrianinsaval, 2024-09-14:
in rare instances I have seen this issue pop up again a few times, and just now I see that rexbas had submitted a fix in coin: https://github.com/coin3d/coin/pull/509 so hopefully this is not going to show up at all anymore when using coin 4.0.3

luzpaz, 2024-09-14:
@adrianinsaval does the appimage run coin4.0.3 yet ?

adrianinsaval, 2024-09-14:
not yet, but I'm looking into it: https://github.com/conda-forge/coin3d-feedstock/pull/32

adrianinsaval, 2024-09-15:
weeklies are now using 4.0.3, I will update RC1 soon too

PR #9699 (2023-06-01): Fix clipping near camera in orthogonal view
Fix #6836 by moving camera backwards the same length as the height of view port.

Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [X]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid  [... cut]
commit 3b92ebea (2023-06-01): Fix clipping near camera in orthogonal view

Fix #6836
PR #11279 (2023-11-20): Gui: Fix window center rotation mode and orthographic view clipping
This PR fixes #11243 and re-fixes #6836. #9699, The original fix for #6836 was a huge improvement but I was still able to reproduce the cut corners and sometimes random glitches. With this fix that does not happen anymore, at least during my testing.

The new approach finds a bounding sphere of the scene, then it is repositioning the camera such that the `nearDistance` can be 0 and the `farDistance` the diameter of the bounding sphere. Finally, to fix #11243, we change the focal distance so the focal point stays the same.

Cur [... cut]