FreeCAD/FreeCAD #6849: [Bug] LeadInOut dressup created on Cancel

Russ4262, 2022-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?style=4&f=15&t=68521

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28793 (Git)
Build type: Release
Branch: master
Hash: 72e91dbbca9094d9bd56b1d47a1eeaaeb7266849
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, OCC 7.5.0
Locale: English/United States (en_US)
Installed mods: 
  * FC_SU
  * Z_MacroStartup
[/code]
```

### Subproject(s) affected?

Path

### Issue description

The LeadInOut Dressup creates the dressup upon activation even when the Cancel button is pressed in its task panel.

Steps to reproduce:
1.  Open a model file, and open Path workbench.
2.  Select the model, and create a Job object with defaults accepted.
3.  Create Profile object with defaults accepted, no feature selection (Base Geometry).
4.  Right-click on Profile in object tree and select LeadInOut Dressup.  (_Or, select Profile in object tree and select the dressup from the Path menu_.)
5.  Click the Cancel button in the LeadInOut task panel.
6.  The dressup is still created and added to the object tree as if OK was pressed.

[PartDesignExample_2.zip](https://github.com/FreeCAD/FreeCAD/files/8646229/PartDesignExample_2.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-07-03:
@Russ4262 is it time to revisit this?

sliptonic, 2024-05-07:
LeadIn/Out could use some refactoring.  It's not terribly complicated.  This would make a good first issue.

closed 2025-01-12 by github-actions[bot]