FreeCAD/FreeCAD #6856: CMake: Quarter support on macOS

luzpaz, 2022-05-09:
Is this FIXME still relevant ?

https://github.com/FreeCAD/FreeCAD/blob/5d49bf78de785a536f941f1a6d06d432582a95d3/src/Gui/Quarter/CMakeLists.txt#L62

luzpaz, 2022-05-09:
@ipatch care to weigh-in ?

ipatch, 2022-05-10:
ill try and get back to you on that one tomorrow. i've been working towards getting python 3.10 support going with macos. i got pyside, shiboken, and coin, but ran into some issues bulding pivy using the cmale setup, need to open an issue about that. and then build a new app bundle hopefully around python 3.10.

life caught up with me today so didn't get troubleshoot my pivy compliation issue.

ageeye, 2022-05-10:
@ipatch Please use the head:

`brew install --head %your%/freecad/pivy`

ipatch, 2022-05-11:
@ageeye 

here's what i'm running into with **pivy**

https://github.com/coin3d/pivy/issues/96

donovaly, 2022-05-15:
What is the status of this bug? If there will not be a PR to fix this in the next week, the bug cannot be fixed for FreeCAD 0.20.

luzpaz, 2022-12-29:
bumped upstream coin ticket

luzpaz, 2023-06-01:
Upstream ticket is dormant. How should we proceed ?

adrianinsaval, 2023-06-01:
The question is, does it affect our usage? If not, I don't think we should care

luzpaz, 2023-06-01:
Closing since this an upstream issue.

closed 2023-06-01 by luzpaz