FreeCAD/FreeCAD #6864: [Bug] Path 3D surface line pattern doesn't cut full width

marshr32, 2022-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=68335

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.28567 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: English/New Zealand (en_NZ)
Installed mods: 
  * A2plus
  * CurvedShapes
  * Curves 0.4.2
  * sheetmetal 0.2.49
[/code]
```

### Subproject(s) affected?

Path

### Issue description

Detailed description and pics in the linked forum post, but briefly.....

When a cut is generated using the line pattern it will produce a part that is over or under dimension depending on the stepover value that's selected vs the width of the cut. 
It will not produce a final path in the series that's less than the specified stepover to cut to the exact dimension.
The 2D pocket tool does produce this last path with smaller stepover, so I think it's a bug in the 3D surface code.

### Anything else?

In this case the stepover on the finish op is set to 13% and the total width of the cut is 1.4mm smaller than it should be.
If the stepover is changed to 12% it's nearly perfect, but it takes a lot of playing to find the correct stepover value manually vs how it's done with 2D pocket where a path with small stepover is added automatically.

[stepover example.zip](https://github.com/FreeCAD/FreeCAD/files/8665310/stepover.example.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-05-10:
Please x-post this ticket to the forum so users know this is getting tracked.

jbaehr, 2024-12-14:
3D surfacing is indeed something I want to look into -- I need it for an own project as well.
However, right now I have way too much on my plate already, sorry.

sliptonic, 2026-10-15:
I will close this issue as not planned. Here is the reason.

The Line and ZigZag patterns make a 2D grid of passes with a fixed stepover, then project the grid onto the model. A projected pattern cannot add a narrower last pass at each edge of the area, because the edges of most models are not parallel to the pattern. Pocket can do this because it offsets each pass from the boundary. A projected pattern does not use the boundary. @Dimitris75 explains this above, with examples.

The new Planar Surface operation replaces 3D Surface and Waterline for planar work. For a model like the one in this issue, with vertical walls that must be cut to size:

- Use Planar Surface with the Waterline or Z-Level Hybrid strategy. These strategies follow the walls at each Z level.
- Or, cut the walls with a Profile operation after the surface pass.

If Planar Surface with one of these strategies does not cut a wall to size, please open a new issue with the file.

closed 2026-10-15 by sliptonic