FreeCAD/FreeCAD #6869: [Bug] SVG Import of rotated group misplaces the group

Kilill, 2022-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=48213

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.27422 (Git)
Build type: Release
Branch: master
Hash: d938733eaf2c2ce7cb18d1cbb56147185c473530
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

Importing an svg that contains grouped elements where the group as been rotated, misplaces the group.

Adding two svg files from created from inkscape one of wich contains grouped elements (fail.svg) the other the elements have been ungrouped after rotation (fail_ungrouped.svg).

The ungrouped version imports correctly the grouped does not. 

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2022-05-11:
Offending code at https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/importSVG.py#L1589

donovaly, 2022-05-14:
What is the status? Is the issue confirmed and is there a fix insight?

chennes, 2022-05-14:
The if statement is checking to see if a rotation center was specified. If it was, then the object is moved so that point is the origin, then rotated, then moved back. I'm away from my computer for a couple of days here, I can look at this on Monday.

donovaly, 2022-05-14:
Thanks Chris. FC 0.20 is still open for fixes for at least a week.

closed 2022-05-16 by chennes

luzpaz, 2022-05-16:
Thanks @chennes!

j-wiedemann, 2024-07-25:
Hello, the bug appears still to occurs. 
Adding reversed to the transformre.findall(tr) list seem to fix this.
[https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/importSVG.py#L1619](https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/importSVG.py#L1619)
`---for transformation, arguments in transformre.findall(tr):`
`+++for transformation, arguments in reversed(transformre.findall(tr)):`