FreeCAD/FreeCAD #6870: [Feature Request] New Datum Line based on Intersection between two Planes

RickX56, 2022-05-11:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?t=62149

### Subproject(s) affected?

PartDesign

### Idea description

Hi, 
I'm requesting an old feature discussed on FreeCAD forum (https://forum.freecadweb.org/viewtopic.php?t=62149)
Would it be possible to create a Datum Line from the intersection of two datum planes? 
There is a macro to do this ([datumplane_intersection.FCMacro] (https://forum.freecadweb.org/download/file.php?id=164282)) but it would be best for it to be imbedded into Part Design Workbench, thus making it parametric.

Thanks for your hard work!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-05-11:
CC @mwganson (as the forum shows your participation)

luzpaz, 2024-02-10:
cc @mwganson could you perhaps test the proposal by LemonBoy ? :point_up_2:

mwganson, 2024-02-11:
I am not able to test at this time.

PR #12328 (2024-02-13): PD: Implement 1D intersection for attachments
Extract a line from the intersection of two faces.

Closes #6870

Please give it a spin, I didn't manage to break it after some use...
commit 7f5d89fa (2024-02-13): Part Attacher: Implement 1D intersection for attachments (#12328)

* PD: Implement 1D intersection for attachments

Extract a line from the intersection of two faces.

Closes #6870

* fixup! Use Handle macro

For some reason I don't fully understand yet the CI was failing, let's
see if this fixes the compilation errors.

* fixup! Add missing(?) header

Well, it compiles just fine on my system...

* fixup! Be more specific in error messages
closed 2024-02-13 by adrianinsaval