FreeCAD/FreeCAD #6875: Preference - Display - Navigation (zoom step)

Kuzma30, 2022-05-11:


luzpaz, 2022-05-11:
Transferred to FreeCAD/FreeCAD for quick discussion

kaktusus, 2022-05-11:
Aleks how did you achieve that the function buttons at the bottom of the window have their labels translated?

[image]

Your point is valid, I didn't even pay attention to it before.
The label is truncated, the caption unreadable.

Kuzma30, 2022-05-12:
Like this? :)
[image]

You can test with this file https://github.com/Kuzma30/FreeCAD/commit/e4c09e15d097b5ab629392bfd7467df49abfab1e

kaktusus, 2022-05-12:
8-O

kaktusus, 2022-05-12:
Where should I copy this file?

kaktusus, 2022-05-12:
[image]

well, I copied it to the standard file location for translations. It worked.
I made one small change in it to make the file 100% compatible with the source file. 
Now there are not even warnings.

What needs to be done to make the change permanent and available to all?

Kuzma30, 2022-05-12:
https://github.com/qt/qttranslations
Branch 5.15.2 more info later if need

donovaly, 2022-05-12:
There is now place also for longer translations:
[image]

commit 354052e8 (2022-05-12): [Gui] Navigation preferences: fix #6875

- as requested by the translators, give a label more space
(the massive changes in the UI file could not be avoided despite the change is just to put the label into another layout column)
closed 2022-05-12 by donovaly