FreeCAD/FreeCAD #6876: [Feature Request] Path: Support G41 and G42

JanneKro, 2022-05-11:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=68467&sid=a06e4def8efce97029d6b00a6154fc11

### Subproject(s) affected?

Path

### Idea description

Professional controllers/workflow use paths as actual center-, outer-, and inner-lines.

There are two reasons for this:

1. It makes it straight-forward to convert a drawing into a program, as all one needs to do is to specify pockets/slots, contours and holes without any conversion, by defining the path and telling the machine which tool to use and on which side of the line to work on (or on the mark in case of holes or tool-wide slots), without having to calculate any offsets.

2. This means that regardless of if gcode has been written by hand or generated through CAD/CAM, the final machine/tool-specific tweaking is left up to the machine [-operator] at any given run. If the fit of the part isn't perfect (due to tool wear, for example), the tool-data can be tweaked in the machine and the same program can be run again, without the need for a new version of the code.

The proposal is thus that Path produces paths on the actual edges of features (without tool compensation) and uses the programmed tool information merely for calculating the number of offset paths, feedrate calculations based on chip-load (if such exist), and instead specify these paths along with G40, G41 or G42 so that the exact offset can be decided by the machine at run-time [or modified by a PP in case G41/G42 aren't supported **(*)**] .

**(*)** I could even see extensions to gcode-senders like bCNC (grbl) to allow an 1/4in endmill to be swapped with a 6mm and still get the same result, if the sender implements this G41/G42 offset before sending the command.
And if not, it would be the job of the PP to perhaps query Path with "can I get this path with this tool radius as offset, please?".

Alternatively, a global setting in Path to turn "machine compensation" on/off.
machine compensation on: Create paths exactly on the features + G40/G41/G42
machine compensation off: Like today.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2022-07-15:
I'm thinking that the profile operation should have a 'compensation type' property. It would select between:
CAM (default): Compensation is done like today using the tool diameter to offset
CONTROLLER: Toolpath would be generated 'on the line' and a G41/2 would be emitted to tell the control to do the offsetting.
HYBRID: Toolpath would be offset using the tool diameter like CAM but a G41/2 would also be issued. Then the programmed offset in the control can be used for wear compensation.
DYNAMIC: Like HYBRID but a G41.1 or G41.2 would be issued along with a D-word for a wear number if it exists as a property in the Toolbit.

silopolis, 2022-07-17:
Hi

Le ven. 15 juil. 2022 à 22:04, sliptonic ***@***.***> a
écrit :

> I'm thinking that the profile operation should have a 'compensation type'
> property.
>
Not only profile I hope !
In front of my mill, I wish all ops would use TLO and CRC so I can easily
tune the job at hand. This was actually one of the frustrating things in
F360: compensation wasn't available in all strategies :/
Chamfering, boring, actually any strategy usable at finishing step must
provide CRC!

> It would select between:
> CAM (default): Compensation is done like today using the tool diameter to
> offset
> CONTROLLER: Toolpath would be generated 'on the line' and a G41/2 would be
> emitted to tell [...]
Looks very good to me as potentially covering all cases and practices,
using tool diameter or 0 diameter...

sliptonic, 2022-07-17:
I tried implementing what I had in mind.  It's not done but I created a draft PR so it's easier to build if you want to try it.
https://github.com/FreeCAD/FreeCAD/pull/7205

To elaborate on what I wrote above.  Regardless of the controller, the user either wants the CAM to calculate the offset of the tool, or he wants the controller on the machine to do that.

There is, of course the other possibility that he wants BOTH.  I've read about different ways of using tool compensation and this is sometimes called the 'west coast' style.   The toolpath is programmed in CAM with offset and the machine's compensation is used just for fine tuning.  This contrasts with 'east coast'  where the toolpath is programmed 'on the line'  and the machine compensation uses the full tool diameter/radius.  

For regular tool compensation, LinuxCNC (and I assume others) use G41/2  with an optional D word.  The D word indicates which tool number in the table to take the offset value from.  In other words the D word is NOT the offset, just the tool number.  Without the D word, the currently loaded tool is used.

LinuxCNC now offers dynamic compensation.  G41.1 / G42.1  This also takes a D word but here the D value IS the offset.
This draft allows for both styles and both types of control.

sliptonic, 2022-07-17:
> Not only profile I hope !
I don't understand what it would look like for the controller to compensate pocketing style operations.  Since the diameter of the cutter is necessary to know how much to offset for a pocket. 

Yes, other kinds of line-following like chamfer.

silopolis, 2022-07-17:
Le dim. 17 juil. 2022 à 16:36, sliptonic ***@***.***> a
écrit :

> To elaborate on what I wrote above. Regardless of the controller, the user
> either wants the CAM to calculate the offset of the tool, or he wants the
> controller on the machine to do that.
>

👍

There is, of course the other possibility that he wants BOTH. I've read
> about different ways of using tool compensation and this is sometimes
> called the 'west coast' style. The toolpath is programmed in CAM with
> offset and the machine's compensation is used just for fine tuning.
>

This is the 0 diameter wear only scenario.

This contrasts with 'east coast' where the toolpath is programmed 'on the
> line' and the machine compensation uses the full tool diameter/radius.
>

This is the actual tool diameter + wear scenario.

For regular tool compensation, LinuxCNC (and I assume others) use G41/2
> with an optional D word. The D word indicates which tool number in the
> table to take the offset value from. In other words the D word is NOT the
> offset, just the tool number. Without the D word, the currently loaded tool
> is used.
>

Actually, this is a compensation data id which, in basic cases, correspond
to the tool ID. Nevertheless, more advanced controls can manage several
compensation data sets for the same tool. For example, Sinumerik handles up
to 9 "cutting edges" per tool. This is very handy when you have to adjust
compensation differently for several ops of the same tool in a single
phase. So with such control you can call 'M6 T1 D2'.

We're having an interesting discussion about this on linuxcnc-devel ML
about this as work is being done on an improved tool database feature by
@TurBoss from the QtPyVCP project.

>
This draft allows for both styles and both types of control.
>

What else?! 😉

JanneKro, 2022-07-17:
Spontaneously, I'm against the idea of having multiple "modes" if we can solve the issue by more aligned means (but still allowing for some flexibility in the end).
The complexity grows quickly in terms of getting every mode to do the right thing in every situation, every change would wherever (dress-ups) need to be confirmed to work for "workflows" that may or may not be used.

Is it not possible to fixate the super-set (G40/41/41.1/42/42.2) as the one and only "FreeCAD-way", and provide clear instructions/helpers on how to create PPs that result in the same...end result?
This way,  even the built-in "simulator" should/could work according to this complete set as a "professional machine". A correctly working PP should thus, if implemented correctly produce the same output as the simulator does.

The way I see it is that FreeCAD's tool library (I don't have the terminology down, sorry), should consist of "sets" (a 'set' being typically something that is loaded in an automatic toolchanger at once / a set that is being used for a particular job (in the case of manual changing)).
Each set contains the "fixed tools" from some tool-dictionary, and the particular wear value for that individual specimen in that set.

Now, when FreeCAD calculates the paths, it will always _only_ care about the tool dimension (NOT the wear) to calculate proper offsets (and number of passes, like in a pocket). 

In a professional machine that has tool wear built-in, the PP will only pass the tool/set information, and the controller will deal with tool diameter and/or wear for that particular tool.

In a "semi-professional"-machine, the PP may have to use the 'wear'-rate from the FreeCAD information, because the controller does not support it.

In a "hobbyist"-machine, the PP will have to ask FreeCAD for offseted-paths (?) to compensate the tool paths at the time of the gcode-generation.

In each and every case, regardless of controller, the end result will (or is supposed to) be exactly as simulated in FreeCAD.
The workflow is the for everyone, the only thing that differs is whether or not someone has to change the wear in FreeCAD and regenerate the gcode, or if it can be done directly in the controller.

Does this make sense to you guys?

JanneKro, 2022-07-17:
> 

It may have been answered in my above post, but I would assume that in order to calculate the number of growing/shrinking paths within a pocket, you would need the tool-diameter and the overlap to do so? What differs here is where the paths start/end (at the edge of the pocket or not).

silopolis, 2022-07-17:
> Not only profile I hope !
>
> I don't understand what it would look like for the controller to
> compensate pocketing style operations. Since the diameter of the cutter is
> necessary to know how much to offset for a pocket.

Which controller? Machine's one? It just follows what CAM tells it to do!?
🤔

Of course the roughing/recessing part of the path, no compensation is
needed. But for finishing, ie the last(s) contours, compensation is indeed
needed... Unless pocketing is a roughing only strategy that must be
followed by an interior profile finishing pass which can rapidly become a
burden and a source of errors.

Yes, other kinds of line-following like chamfer.

Typically one of the ops where compensation was unavailable in f360 🤬
leading most to use contouring strategy instead 😒

JanneKro, 2022-07-17:
> To elaborate on what I wrote above. Regardless of the controller, the user either wants the CAM to calculate the offset of the tool, or he wants the controller on the machine to do that.

@sliptonic Do you count "PP" to be a part of "CAM"? 
Should it matter to "the user"?  The user wants to create a pocket, and the gcode should be correct for the machine.
If the CAM only has options in operations for "On the line (G40)" or inside/outside (G41/42), then the CAM should always produce those, and the controllers that do not handle them need PPs that change these to hardcoded offsets from the tool-data.

silopolis, 2022-07-17:
Le dim. 17 juil. 2022 à 19:37, JanneK ***@***.***> a écrit :

> Spontaneously, I'm against the idea of having multiple "modes" if we can
> solve the issue by more aligned means (but still allowing for some
> flexibility in the end).
>

I don't think you can avoid multiple modes...

The complexity grows quickly in terms of getting every mode to do the right
> thing in every situation, every change would wherever (dress-ups) need to
> be confirmed to work for "workflows" that may or may not be used.
>

But the situation is not that bad because there are really only two path
computation modes!
First one is offset by tool diameter (theorical or actual), ie path
controls tool center.
Second follows part/feature contour, so called "on the line", and path
controls tool contact point.

Then, wether and which comp type should be output is just an option read
and interpreted by PP:
First mode, no comp: G40 in program and/or op header and that's all;
programer uses actual tool dimension in CAM, control has no clue.
First mode, comp: G41/G42 before compensated path, G40 at the end; CAM uses
tool dia, tool dia is 0 in control and control only uses wear.
Second mode, comp: G41/G42 before compensated path, G40 at the end ; CAM
uses contact point, control uses tool dia (not 0)+wear, ie actual tool
radius.

"Dynamic" mode is just a variation of the last one on the way the radius
data is obtained.

Is it not possible to fixate the super-set (G40/41/41.1/42/42.2) as the one
> and only "FreeCAD-way",
>

It is, and not only FC's one 🙂
But you have to provide the 4 "modes" to cover the different situations and
work habits.

and provide clear instructions/helpers on how to create PPs that result in
> the same...end result?
>

I don't really get what you mean here 🤔

> PP receives path points/move that it translates into G-Code moves,
corresponding to either the offset or the on the line path, optionally
wrapped into comp start/stop messages that it tranlsates into G41*/G42* and
G40.

It's the CAM programmer job to choose between those 4 universal scenari, ie
the one suited among those supported by his control, and the PP's one to
output it in the specific machine idiom.

This way, even the built-in "simulator" should/could work according to this
> complete set as a "professional machine". A correctly working PP should
> thus, if implemented correctly produce the same output as the simulator
> does.
>

This is a must indeed

Each set contains the "fixed tools" from some tool-dictionary, and the
> particular wear value for that individual specimen in that set.
>
>
See my other message for other, more advanced cases.

Now, when FreeCAD calculates the paths, it will always *only* care about
> the tool dimension (NOT the wear) to calculate proper offsets (and number
> of passes, like in a pocket).
>

CAM indeed does not not handle wear per se -BUT for machines without CRC,
programmer will have to adjust tool dimension and regenerate to take wear
into account.
Wether to offer a dedicated field for this case is more a UX choice. Of
course, supporting wear data in tool library would be nicely beneficial for
those users as this would ease reuse of wear data between jobs.

In a "hobbyist"-machine, the PP will have to ask FreeCAD for offseted-paths
> (?) to compensate the tool paths at the time of the gcode-generation.
>

All path computations should happen in CAM. PP is "just" an advanced
translator between generic PP input commands and control specific language.
Thus, provided the simulation is accurate, the very same CAM path, ie same
set of intermediary commands, post-processed by specific PPs for different
machines should deliver identical parts, true to simulation.

In each and every case, regardless of controller, the end result will (or
> is supposed to) be exactly as simulated in FreeCAD.
>

Absolutely

The workflow is the for everyone, the only thing that differs is whether or
> not someone has to change the wear in FreeCAD and regenerate the gcode, or
> if it can be done directly in the controller.
>

Yep.
I'd add that for the dynamic scenario, it would be nice to have the option
to output the D address value as a parameter. Operators will be grateful 😉

TY
J

CasualObserverofCode123, 2025-01-17:
This new FreeCad 1.0 is awesome, but the removal of the `UseMachineCRC` (G41/42) property in `LeadInLeadout` feature after FreeCad 020.2 is a frustruating omission (formerly available in 0.20.2 and subsequently removed) from an extremely powerful and almost complete CAD/CAM software suite. Without sounding ungrateful or unappreciative of the work that the devs put forth on this, I have to ask if there is a timeline on the return of this feature or if it could be elevated to a higher priority? Or, in the interim is there a set of clear instructions for a workaround in this newest version?

Currently, I need the feature badly enough whereby I have resorted to modifying the G-Code manually (extremely error prone) or loading my files back into FreeCad 0.20.2 to utiliize the `UseMachineCRC` feature. Any assistance for a temporary workaround/workflow would be greatly appreciated. I want to close by saying others on various social platforms echo a similar confusion as to why this feature was abruptly removed and not replaced. Again, I appreciate the incredible work the devs do, but I think the frustration this redditor has mirrors my own: [see this comment](https://www.reddit.com/r/FreeCAD/comments/1fdqkhl/comment/m0omc6h/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)

rmu75, 2025-01-17:
As a stop-gap measure you could use this ugly hack https://gist.github.com/rmu75/03a677b80debdeee49cc8476c2894fc4

Be careful, no compensation and extra distance set to zero in CAM Profile may reverse cut direction depending on geometry and phase of moon, best use some small positive extra distance like 1µm.

I'm not sure what a proper implementation of G41/G42 feature would look like (and where it would belong), the "CW/CCW" vs. "climb/conventional" situation and clipperlib has to be sorted first IMHO.

CasualObserverofCode123, 2025-01-18:
Thanks for the quick reply. I'll give this a try and come back here to let you know how it all goes. 

Have had predictable results with the feature in 0.20.2 for simple geometries but will keep an eye out for unusual behavior.

Thanks again for jumping in to help with this.

luzpaz, 2025-06-01:
Any proposals ?