FreeCAD/FreeCAD #6878: [Part] BOPTools/ShapeMerge.py: code appears to indicate it's inefficient

luzpaz, 2022-05-12:
https://github.com/FreeCAD/FreeCAD/blob/5d49bf78de785a536f941f1a6d06d432582a95d3/src/Mod/Part/BOPTools/ShapeMerge.py#L100

AjinkyaDahale, 2022-05-23:
We can replace
https://github.com/FreeCAD/FreeCAD/blob/5d49bf78de785a536f941f1a6d06d432582a95d3/src/Mod/Part/BOPTools/ShapeMerge.py#L99-L102
with
```
l_groups = len(groups)
groups = [groups[i_group] for i_group in connected_to if i_group < l_groups]
```
This removes the unnecessary search in `connected_to`

AjinkyaDahale, 2022-12-31:
I can make the PR, but I need someone to back up what's happening.

luzpaz, 2022-12-31:
@DeepSOIC this is code you added. Could you lend a moment of your time to weigh-in? :pray:

DeepSOIC, 2023-01-01:
I don't quite remember why i thought it's inefficient. The odds of connected_to being long are pretty low - I can't think up an example of how it can be longer than 2 for manifold shapes (except compsolids, maybe). So, i think it is safe to remove this comment. On the other hand, i think it may be possible to improve the performance of the whole algo significantly... i'm actually having a hard time even estimating the O() complexity of it, because these aren't just plain random lists - topology is involved.

AjinkyaDahale, 2023-01-01:
@DeepSOIC thanks for the input. I had still made this: https://github.com/AjinkyaDahale/FreeCAD/tree/part-fix-6878. Probably an overkill for the list sizes you mentioned. We can discuss further what we can do instead.

luzpaz, 2023-11-26:
@AjinkyaDahale what's the fate of this ticket ?

AjinkyaDahale, 2023-11-26:
> @AjinkyaDahale what's the fate of this ticket ?

Let's close it. There may be need for a larger scale overhaul but this one snippet is already changed and merged.

closed 2023-11-26 by luzpaz