FreeCAD/FreeCAD #6879: [Draft] draftguitools/gui_fillets.py code apparently needs to be modified to Qt5 style

luzpaz, 2022-05-12:
https://github.com/FreeCAD/FreeCAD/blob/c9b772cf7d0ca5ee6bd899ada3922f6d5452522d/src/Mod/Draft/draftguitools/gui_fillets.py#L98-L104

donovaly, 2022-05-14:
Can you create a PR that fixes this?

luzpaz, 2022-05-14:
I don't know what that 'style' is

donovaly, 2022-05-14:
> I don't know what that 'style' is

Can you please bring this to the developer forum and CC @Roy-043 . I guess Roy knows more.

luzpaz, 2022-05-15:
Thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=68703

Roy-043, 2022-05-15:
I know very little about Qt. So I can't be of assistance here. Note that similar code is used in other sections of the Draft code.

donovaly, 2022-05-15:
Thanks @Roy-043 , inviting @wwmayer hoping he can state if the style is OK or the Todo in the code is wrong.

yorikvanhavre, 2022-05-16:
i think qt5 style means: instead of `OtCore.QObject.connect(widget,signal,slot)`, use `widget.signal.connect(slot)`. But it's rather cosmetic only..

donovaly, 2022-05-17:
>i can do this tomorrow (out of home today)

Thanks.

closed 2022-05-17 by yorikvanhavre

luzpaz, 2022-05-17:
Thanks @yorikvanhavre should we open a ticket for other instances of this code formatting? Or do these functions eventually get overridden by the necessity to create .ui files for such files? I ask because I just noticed https://github.com/FreeCAD/FreeCAD/commit/27277e29c68d3f9b2f0d10b70de1ca8444868be5#diff-912c746bfbadeb6a064de2781924e243b63ab3b039376e532c83113d2037bab8R25-R28

```text
TODO: Currently this tool uses the DraftGui widgets. We want to avoid using
this big module because it creates manually the interface.
Instead we should provide its own .ui file and task panel,
similar to the OrthoArray tool.
```

yorikvanhavre, 2022-05-17:
I would say these TODO comments are not all meant to be addressed immediately... For ex. this one is mostly meant at the next person touching that code, if they have courage to undertake some cleaning :) Honestly, to not pollute the tracker too much, I wouldn't turn them all into issues. Everything is still perfectly fine if nobody does it.

If you spot one that looks like it would really need to be done rather soon, of course, then it's a different story...

luzpaz, 2022-05-17:
Understood. JFYI, I've been using that logic to an extent already because I've been trying to be mindful of not polluting the tracker but at the same time I have found some FIXME's and TODO's that might merit it. But back to my question above, what I surmise is that the overall intention eventually is to move away from generating .ui widgets in code to create dedicated `.ui` fies for files that are?

yorikvanhavre, 2022-05-17:
Basically the old DraftGui code creates all kinds of widgets "by hand", without using .ui files. Since that file is an old dinosaur hard to modify, the idea is to stop using it wherever possible, for example for all commands that can have their ui done in an own .ui file. But it's not always simple, because DraftGui has a lot of additional functionality tied into its widgets

commit 27277e29 (2022-05-17): Draft: Fixed qt connect style - fixes #6879