FreeCAD/FreeCAD #6884: [Feature Request] Support to create holes from sketch points

wwmayer, 2022-05-15:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=68696

### Subproject(s) affected?

PartDesign

### Idea description

Currently a circle must be added to a sketch to allow to make a hole afterwards. Since the radius of a circle is ignored anyway it would be convenient to just add points and use them for the holes.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

zackwhit, 2022-09-02:
I've been working on this a bit, gotten a draft of the feature working. I will update my fork with the changes once I can refine them a bit. Nobody else has already fixed this, right? lol

[image]

[learning notes as I go](https://www.zackwhitlock.com/post/freecad-devlog-2-5-points-as-a-basis-for-holes)

pierreporte, 2023-10-28:
I commented in https://github.com/FreeCAD/FreeCAD/issues/10100#issuecomment-1783929916 that it could be useful to have two kinds of points for construction geometry and regular geometry. It would allow to build complex patterns with a lot of geometry and only make visible from outside the sketch the position of holes using regular points.

pierreporte, 2024-01-06:
I opened an issue (#11911) to implement normal points in sketches (currently there are only construction points). They are a requirement to implement this feature request.

maxwxyz, 2024-03-02:
Remarks from the FreeCAD Day 2024 Complaint Session:
I've got an unconstrained sketch and then the need to draw circles of arbitrary sizes.
We need the sketch to say where your holes are going to go, but you shouldn't need to draw a circle. You should have something like a point or location/layout.
You want to say there are features in these five places. And then you go into the hole tool and it will put the features in those places. And currently you just draw a circle of arbitrary. So you only draw the circle to have the point in the center.

maxwxyz, 2024-12-25:
@zackwhit are you still working on this?

zackwhit, 2024-12-26:
> @zackwhit are you still working on this?

No. I put it all on pause while waiting for the TNP merge; no idea where it stands now.

maxwxyz, 2024-12-27:
@zackwhit the TNP mitigation code has been merged and is in the stable 1.0 version. Are you planning to pick this up again?

zackwhit, 2025-01-01:
> @zackwhit the TNP mitigation code has been merged and is in the stable 1.0 version. Are you planning to pick this up again?

Not planning on picking it up again; at least anytime soon.

theo-vt, 2025-03-05:
Hey! this has been bugging me as well and I thought I would give it a try,

I was wondering if it would be doable to add a point type to the sketcher to differentiate between a "curve handle" and a "point"

[image]

Inventor has little spikes on points explicitly added by the user [left] by opposition to curve handles [right] and I think it looks neat. The hole tool could target all explicit points by default in addition to circles/arcs and the selected points otherwise?

thanks,

pierreporte, 2025-03-06:
@theo-vt There is already a dedicated point type (the point tool in sketcher), a proposal to use it better than now (see https://github.com/FreeCAD/FreeCAD/issues/13391#issuecomment-2585909362), and a proposal for how everything should look (in the design working group space on the FreeCAD Discord server).

theo-vt, 2025-03-07:
Thanks for the pointers, I'll look into that!

theo-vt, 2025-03-12:
I am not sure how to approach the problem and am wondering if it would make sense to modify the profile before or as soon as it gets to the Hole class such that points are transformed into circles if no circle/arc is found in the profile? After that the normal hole code could be kept exactly the same,

pierreporte, 2025-03-12:
The problem with this approach is that there will not be any error or warning if no defining point is found in the sketch. IMO the best way is to check the version of the document file, and act differently if it's older that the version the feather was merged.

pierreporte, 2025-03-26:
IIRC @kadet1090 has experience with migrations.

closed 2025-04-21 by kadet1090

K4LCIFER, 2025-09-09:
What release was this feature added in? I don't see it in FreeCAD v1.0.2, but given that this was merged in April, I feel like it should be there.