FreeCAD/FreeCAD #6885: [Bug] PartDesign Thickness tool contains unimplemented options

FEA-eng, 2022-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=19&t=56332

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.28774 (Git)
Build type: Release
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

PartDesign Thickness tool offers the selection of three modes, only one of which (Skin) works. The Pipe and Recto Verso modes are not implemented and thus do nothing. What's more, the Intersection checkbox shouldn't be used since its enabled state is also not implemented properly.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

0penBrain, 2022-06-08:
This ticket should make things clear :
 * FreeCAD correctly implements all possibilities offered by OCC
 * Statements given in the issue description are actually OCC missing/incomplete implementations

rahmanshaber, 2022-08-30:
Will you guys ever work on the thickness tool. for 3 hours i am trying to make a shell from a part, still didn't accomplished anything. 

The answer might be open cascade's issue but asking user to deal with open cascade is not a best answer i believe.

luzpaz, 2022-08-31:
> Will you guys ever work on the thickness tool....The answer might be open cascade's issue but asking user to deal with open cascade is not a best answer i believe.  

You're totally missing the point: we're ALL users here. No one has stepped up yet to fix it (not because they don't want to...but) because the complexity is no joke. "*Why isn't this fixed yet?!*" rhetoric doesn't get you far in the FLOSS world unless you're following-up with "How can I help make it happen with the skills that I have?".

rahmanshaber, 2022-08-31:
> You're totally missing the point: we're ALL users here. No one has stepped up yet to fix it (not because they don't want to...but) because the complexity is no joke. "_Why isn't this fixed yet?!_" [...]

I know how FOSS apps work, i run 2 of them, And i wanted to sponsor one guy to fix it, but he said he will let me know when he gets some time to do the work, but mainly he doesn't think it's that important to do work on that. It always comes to what individual thinks about something, doesn't matter what others facing.

luzpaz, 2022-08-31:
> It always comes to what individual thinks about something, doesn't matter what others facing. 

Was going to respond to this with a tongue-in-cheek reply but that's counter-productive. 
Instead I'll say that I'm glad to hear that you tried to rally some support to get thickness addressed. Maybe do so more publicly? (this is the first I'm hearing of your efforts)

rahmanshaber, 2022-08-31:
@luzpaz 

https://forum.freecadweb.org/viewtopic.php?f=3&t=64820
https://forum.freecadweb.org/viewtopic.php?f=3&t=64795&start=40

FEA-eng, 2024-12-14:
Still relevant

codeotaku, 2026-09-01:
Well... I still think we should remove the unimplemented features from the UI.... but, I'm having some (preliminary) success implementing the recto verso algo

[image]

closed 2026-09-08 by maxwxyz