FreeCAD/FreeCAD #6892: [python3.10 conda] <fails to start freecad>

looooo, 2022-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

no forum disussion

### Version

0.20 (Development)

### Full version info

```shell
FreeCAD 0.20 Revision: 28873 (Git)
same issue for osx and linux.
```

### Issue description

```
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 685, in <module>
  File "<string>", line 108, in InitApplications
  File "/home/lo/miniconda3/envs/freecad_py310/lib/python3.10/posixpath.py", line 395, in realpath
    path, ok = _joinrealpath(filename[:0], filename, strict, {})
  File "/home/lo/miniconda3/envs/freecad_py310/lib/python3.10/posixpath.py", line 430, in _joinrealpath
    st = os.lstat(newpath)
ValueError: embedded null byte
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 276, in <module>
  File "<string>", line 131, in InitApplications
<class 'AttributeError'>: module 'FreeCAD' has no attribute '__ModDirs__'
Error in FreeCADGuiInit.py: module 'FreeCAD' has no attribute '__ModDirs__'
```

this only occurs with python3.10.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ageeye, 2022-05-16:
Do you check the content of [LibFcDir](https://github.com/FreeCAD/FreeCAD/blob/master/src/App/FreeCADInit.py#L107)?

`Log(LibFcDir.encode('ascii').hex())`

looooo, 2022-05-17:
Interestingly this issue does not occure on windows:

```
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.28893 (Git)
Build type: Release
Python 3.10.4, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: German/Austria (de_AT)
Installed mods: 
  * FCGear
  * Plot
```

looooo, 2022-05-17:
freecadcmd still works somehow. The output of `FreeCAD.getLibraryDir()` results in:

```
'/home/lo/miniconda3/envs/freecad_py310//home/lo/miniconda3/envs/freecad_py310/lib\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

and 
```
os.path.realpath(FreeCAD.getLibraryDir())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/lo/miniconda3/envs/freecad_py310/lib/python3.10/posixpath.py", line 395, in realpath
    path, ok = _joinrealpath(filename[:0], filename, strict, {})
  File "/home/lo/miniconda3/envs/freecad_py310/lib/python3.10/posixpath.py", line 430, in _joinrealpath
    st = os.lstat(newpath)
ValueError: embedded null byte
```

So it seems there is something wrong with `FreeCAD.getLibraryDir()`

ageeye, 2022-05-17:
I think the trouble is [here](https://github.com/FreeCAD/FreeCAD/blob/master/src/App/Application.cpp#L1142). Is LIBRARYDIR set by cmake? It not contains consistent values:

```
>>> import os
>>> FreeCAD.getLibraryDir()
'C:/Program Files/FreeCAD/FreeCAD_weekly-builds-28751-Win-Conda_vc14.x-x86_64/D:/bld/freecad_1650781659541/_h_env/Library/lib'
>>> os.path.realpath(FreeCAD.getLibraryDir())  
'D:bld\\freecad_1650781659541\\_h_env\\Library\\lib'
```

looooo, 2022-05-17:
uh it's done by cmake... this is for sure a problem. Is there no way to find the library-dir at runtime? The problem is that app images and conda-environments have no constant directory structure.

wwmayer, 2022-05-17:
> Is there no way to find the library-dir at runtime?

No. If the extension modules are installed into /usr/lib/x86_64-linux-gnu then FreeCAD won't be able to find them there by default. See issue #6274 which was fixed with 6bc65bdd99d

Can you check the value of the CMake variable `CMAKE_INSTALL_LIBDIR`? In FreeCADApp the variable `LIBRARYDIR` is used as compiler define. Can you also check the content of the created Makefile and check the value of `LIBRARYDIR`?

The same technique is used to set the resource and doc directories. What happens if you call `FreeCAD.getHelpDir()` or `FreeCAD.getResourceDir()`.

looooo, 2022-05-17:
looks good for both environments:

python3.9:
```
>>> import FreeCAD
>>> FreeCAD.getHelpDir()
'/home/lo/miniconda3/envs/freecad/share/doc/FreeCAD/'
>>> FreeCAD.getResourceDir()
'/home/lo/miniconda3/envs/freecad/share/'
```
python3.10:
```
>>> import FreeCAD
>>> FreeCAD.getHelpDir()
'/home/lo/miniconda3/envs/freecad_py310/share/doc/FreeCAD/'
>>> FreeCAD.getResourceDir()
'/home/lo/miniconda3/envs/freecad_py310/share/'
```

not sure about the appimage.

wwmayer, 2022-05-17:
In some answers at https://stackoverflow.com/questions/32947974/file-path-with-cmake-add-definitions it's recommended to put extra quotes. But since it works for DOCDIR and RESOURCEDIR I am not sure if this is really needed.

But can you also post the value of CMAKE_INSTALL_LIBDIR? Btw, the value passed to the compiler might be in a file called flags.make in the FreeCADApp build directory.

looooo, 2022-05-17:
@wwmayer sry, I haven't compiled freecad for a long time on my local machine. Maybe this helps:
https://github.com/looooo/freecad-feedstock/blob/master/recipe/build.sh#L53

wwmayer, 2022-05-18:
> `-D CMAKE_INSTALL_LIBDIR:FILEPATH=$PREFIX/lib \`

OK, you make it an absolute path here. This explains the weird path name
 'C:/Program Files/FreeCAD/FreeCAD_weekly-builds-28751-Win-Conda_vc14.x-x86_64/D:/bld/freecad_1650781659541/_h_env/Library/lib' ageeye has reported.

wwmayer, 2022-05-18:
f3fd1d2035c4d fixes the issue with the absolute path name. But I don't know if the issue with null bytes is related to that. Can you run a new build and test, please?

wwmayer, 2022-05-18:
Great!

looooo, 2022-05-25:
This is still an issue for osx:
```
>>> FreeCAD.getLibraryDir()
'/Users/lo/mambaforge/envs/freecad_osx64/lib\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

```
>>> FreeCAD.Version()
['0', '20', '28918 (Git)', 'Unknown', '2022/05/21 18:09:51', 'master', 'e1f2685734']
```

luzpaz, 2022-05-31:
@ipatch do you notice this issue on macOS Homebrew ?

looooo, 2022-05-31:
@luzpaz I guess this is conda/py310 specific.

@conda-forge/core is there any related issue?

luzpaz, 2022-05-31:
@looooo keep this ticket here or move to FreeCAD-Conda or perhaps FreeCAD-Bundle ?

ipatch, 2022-05-31:
```
>>> FreeCAD.Version
<built-in function Version>
>>> FreeCAD.Version()
['0', '20', '28901 +34 (Git)', 'git://github.com/ipatch/FreeCAD.git ipatch.builds.0.20beta1', '2022/05/29 21:53:37', 'ipatch.builds.0.20beta1', '8505cf72df8794d39f847939f751b2d0c15c57a5']
>>> FreeCAD.getLibraryDir()
'/Applications/freecads/tmp-create-dmg/FreeCAD.app/Contents/lib'
>>> FreeCAD.getResourceDir()
'/Applications/freecads/tmp-create-dmg/FreeCAD.app/Contents/share/'
>>> FreeCAD.getHelpDir()
'/Applications/freecads/tmp-create-dmg/FreeCAD.app/Contents/share/doc/FreeCAD/'
>>> 
```

same app bundle moved to a different directory location on my local filesystem

```
>>> FreeCAD.Version()
['0', '20', '28901 +34 (Git)', 'git://github.com/ipatch/FreeCAD.git ipatch.builds.0.20beta1', '2022/05/29 21:53:37', 'ipatch.builds.0.20beta1', '8505cf72df8794d39f847939f751b2d0c15c57a5']
>>> FreeCAD.getLibraryDir()
'/Applications/freecads/my.cmake.builds/catalina.build.box/FreeCAD.app/Contents/lib'
>>> FreeCAD.getResourceDir()
'/Applications/freecads/my.cmake.builds/catalina.build.box/FreeCAD.app/Contents/share/'
>>> FreeCAD.getHelpDir()
'/Applications/freecads/my.cmake.builds/catalina.build.box/FreeCAD.app/Contents/share/doc/FreeCAD/'
>>> 
```

looooo, 2022-06-08:
answer from @chrisburr via gutter:

> I doubt it's Python 3.10 specific, by default conda does binary relocation by using a long directory name at build time and then padding with null bytes at install time
It works for most strings but if the string isn't null terminated you get this
The sources probably need patching so it's treated as a null terminated string

@wwmayer is it possible that we check for null terminated string?

ageeye, 2022-06-08:
I can test it only with taps, because I do know how get null chars from cmake:

```
cmake_minimum_required(VERSION 3.13)
project(test_example)
set(CMAKE_CXX_STANDARD 17)

set(LIBRARYDIR "Blupp\t\t\tE")
add_definitions(-DLIBRARYDIR="${LIBRARYDIR}")

set(SOURCE_FILES main.cpp)

add_executable(test_example ${SOURCE_FILES})
```

```
#include <iostream>

int main() {
    std::string path(LIBRARYDIR);
    path.erase(std::remove(path.begin(), path.end(), '\t'), path.end());
    std::cout << path << std::endl;
    return 0;
}

```

ageeye, 2022-06-09:
@looooo Is it possible to build a PR with Conda?

looooo, 2022-06-09:
Not sure what you mean. If you can provide a patch (git diff) I can easily test this with the freecad-feedstock.

wwmayer, 2022-06-09:
> answer from @chrisburr via gitter:

I agree insofar that it's rather a Conda issue than Py3.10 related as otherwise we should see it elsewhere too.

> by default conda does binary relocation by using a long directory name at build time and then padding with null bytes at install time

But how should Conda at this point know that the built-in string is a path? I cannot imagine that Conda can do that but instead I guess that the path is already corrupted at compile time by introducing the `'\0'` characters into the string that is passed to the compiler.

> @wwmayer is it possible that we check for null terminated string?

We could do this but then I have to know how the string looks like because if the passed string is of this form `"AValidPath\0\0\0\0\0\0\0"` then by assigning the C string literal to a std::string the assignment stops after the first null-termination character.

The `\0` would only be kept if the length is passed to std::string but this isn't the case in FreeCAD

```
#include <string>
#include <iostream>

#define LIBRARYDIR "/tmp/\0\0\0\0\0\0\0\0\0_Path"

int main()
{
    std::string path(LIBRARYDIR);
    std::cout << "C++ string: " << path << ", Length: " << path.size() << std::endl;

    std::string path_with_null(LIBRARYDIR, 19);
    std::cout << "C++ string: " << path_with_null << ", Length: " << path_with_null.size() << std::endl;
}
```

The output of this test program is:
C++ string: /tmp/, Length: 5
C++ string: /tmp/_Path, Length: 19

chrisburr, 2022-06-09:
I'm a little short on time and haven't read this issue fully but hopefully I can clear things up a little.

This is an issue that doesn't affect compile time at all. All conda does here is make sure that the install directory path is longer than anyone would reasonably want.

Then when someone calls `conda install` the conda looks inside the binaries for sequences of bytes that:

* contains the install location that was used at build time
* end with a null byte at some point

For example if the build was originally installed in in `/long/path/to/install/prefix` and the code contains:

```c
char* message = "The extra data is in /long/path/to/install/prefix/something.json!!!";
```

Then conda will notice this string in the binary and when installing in `/short/prefix` it'll replace:
```
The extra data is in /long/path/to/install/prefix/something.json!!!\0
```
with:
```
The extra data is in /short/prefix/something.json!!!\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0
```

IIRC the common cause of this failing is something like:

```
const std:string message = "The extra data is in /long/path/to/install/prefix/something.json!!!";
```

Now, because the compiler knows that the string is always going to be 67 characters long it can end up optimising based on assumptions that don't hold at runtime. The simpliest solution is often to add a `.c_str()` in an appropriate place to ensure the string is treated as if it's null terminated rather than a fixed length.

See https://github.com/conda/conda-build/issues/1674 for more details.

wwmayer, 2022-06-09:
> I'm a little short on time and haven't read this issue fully but hopefully I can clear things up a little.

Thanks for the explanation and the link.

luzpaz, 2022-06-09:
@looooo please close when confirmed fix, thanks!

looooo, 2022-06-09:
thanks, yes now it works also on osx
```
[FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]
>>> FreeCAD.Version()
['0', '20', '29085 (Git)', 'Unknown', '2022/06/09 15:40:37', 'master', '7225e5f0ba']
>>> FreeCAD.getLibraryDir()
'/Users/lo/mambaforge/envs/freecad_osx64_py310/lib'
>>> FreeCAD.getResourceDir()
'/Users/lo/mambaforge/envs/freecad_osx64_py310/share/'
>>> FreeCAD.getHelpDir()
'/Users/lo/mambaforge/envs/freecad_osx64_py310/share/doc/FreeCAD/'
```

closed 2022-06-09 by looooo