FreeCAD/FreeCAD #6893: [label translation]  Sketcher → Combo view (panels) → Tasks (tab)

kaktusus, 2022-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://github.com/FreeCAD/FreeCAD-translations/issues/106

### Version

0.20 (Development)

### Full version info

```shell
OS: Debian GNU/Linux bookworm/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.28872 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 9057658)
Hash: 9057658ae5d638f6903bc831b53ceb349c2870a2
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
...
```

### Subproject(s) affected?

Sketcher

### Issue description

In response to donovaly's request, I am posting a description of the issue I discovered.

[image]

1.  `DoF' - The string is present in Crowdin and is marked as hidden and duplicated
    https://crowdin.com/translate/freecad/569/en-pl#6597724

I would suggest adding the plural variety at this point. And remove the current abbreviation.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-05-17:
@abdullahtahiriyo can you please have a look at this Sketcher translation issue?

abdullahtahiriyo, 2022-05-17:
@donovaly 

I am not sure I understand what I am looking at.

The issue is about translation, but the UI is in English language. In English it makes sense to me (but I am not native).

Because the OP is suggesting to provide plural, I am not sure if that is what is intended. The plural of Degree of Freedom is Degrees of Freedom, I am not sure if DoFs should be used. This was actually provided by a native speaker @chennes.

So I think there is a need to clarify what is exactly what is not ok and what should be the solution.

kaktusus, 2022-05-17:
Perhaps I can provide additional clarification:

When I searched for this phrase in Crowdin I used the translated string to search.
The search engine found one occurrence. I made an entry for the issue in Crowdin.

After a while I repeated the search by typing `DoF` this time the search engine found two hits. My report in Crowdin is not relevant.

However, a string variation according to the quantitative value of the parameter would be appreciated.

chennes, 2022-05-17:
Do you want us to provide the `n` argument to `tr()`, or to just split the use of "DoF" into two branches, one for 1 DoF, and one for more than one DoF?

kaktusus, 2022-05-18:
Here is an example to make things clear:

[image]

https://crowdin.com/translate/freecad/27907/en-pl#6589158

Right now it looks bad.
[image]

Kuzma30, 2022-06-07:
@luzpaz 
https://github.com/Kuzma30/FreeCAD/commits/DoF
Can I write DoF(s) for English?
I also change tooltip text "Number of DoFs" and made it translatable

luzpaz, 2022-06-07:
I'm confused. Hard to understand from a mobile browser. Will respond soon

donovaly, 2022-06-19:
I think the issue was fixed for the 0.20 release. If not, what is now missing?

kaktusus, 2022-06-19:
Yes the issue has been solved.

[image]

closed 2022-06-19 by kaktusus