FreeCAD/FreeCAD #6900: [Bug] <title>

Jihadist, 2022-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=66981

### Version

0.19 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

### Issue description

FreeCAD renders long (very long) colored lines as black when importing from svg
FreeCAD view:
[image]
Firefox view:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2022-05-18:
The issue here is that the color info is done on groups, not on geometry. See [file to download](https://forum.freecadweb.org/download/file.php?id=182291). Seems supported by Firefox, and I couldn't see in inkscape... We need to see if this is officially supported in SVG

donovaly, 2022-05-21:
> Seems supported by Firefox, and I couldn't see in inkscape.. We need to see if this is officially supported in SVG

I tried the brand-new Inkscape version 1.2 and also with this, the example SVG from the previous post is not displayed as you expect it - there are no lines at all and Inkscape only allows you to zoom out down to 1.0 % of the original size.

> We need to see if this is officially supported in SVG

It is see
https://www.w3.org/TR/SVG11/struct.html#GElement
and as example
https://developer.mozilla.org/en-US/docs/Web/SVG/Element/g

So independent that Inkscape has issues with the file, it is according to the SVG 1.1 specs and we should therefore support it.

yorikvanhavre, 2022-05-21:
indeed... and the group supports the whole array of styling that normal entities do... okay, indeed we need to add that!

donovaly, 2022-05-22:
I reported the bug also to Inkscape:
https://gitlab.com/inkscape/inbox/-/issues/7030

luzpaz, 2022-12-29:
@donovaly can you bump the inkscape ticket to get more exposure ?

donovaly, 2022-12-30:
> @donovaly can you bump the inkscape ticket to get more exposure ?

What does "bump a ticket" mean?
I reported it correctly and I cannot fix it in Inkscape. The Inkscape guys are also only volunteers.

luzpaz, 2022-12-30:
It simply means to type 'Bump' in the ticket so it brings it back to the Inkscape team's radar

luzpaz, 2023-04-13:
@0penBrain does #9242 address this issue ?

luzpaz, 2023-04-18:
Should this ticket be in the v0.21 release roadmap ?

yorikvanhavre, 2023-05-09:
sample file: [test.svg](https://github.com/FreeCAD/FreeCAD/assets/1136856/1307013d-9583-4408-b11a-ae34cb3772cb)

PR #9522 (2023-05-13): Draft: Support group styles in SVG import - fixes #6900
Styles applied to groups will now be applied to all their children, unless overridden
fixes #6900

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
commit 7e357513 (2023-05-13): Draft: Support group styles in SVG import - fixes #6900 (#9522)

* Draft: Support group styles in SVG import - fixes #6900

* Draft: SVG importer - apply styleto polylines
closed 2023-05-13 by Roy-043