FreeCAD/FreeCAD #6902: conda: test errors

looooo, 2022-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

no forum discussion

### Version

0.20 (Development)

### Full version info

```shell
OS: Ubuntu 18.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.28904 (Git)
Build type: Release
Branch: (HEAD detached at 8ad30d0)
Hash: 8ad30d03146fef0f39b90073fdf3e6072ed7f66d
Python 3.10.4, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator
  * CurvedShapes
  * Beltrami
  * Curves 0.3.0
  * Assembly4
```

### Issue description

testing the gui-tests I run into this error

```
Traceback (most recent call last):
  File "/home/lo/miniconda3/envs/freecad_py310/lib/python3.10/unittest/case.py", line 59, in testPartExecutor
    yield
  File "/home/lo/miniconda3/envs/freecad_py310/lib/python3.10/unittest/case.py", line 591, in run
    self._callTestMethod(testMethod)
  File "/home/lo/miniconda3/envs/freecad_py310/lib/python3.10/unittest/case.py", line 549, in _callTestMethod
    method()
  File "/home/lo/miniconda3/envs/freecad_py310/Mod/Mesh/MeshTestsApp.py", line 447, in testRayPick
    det=coin.cast(det,str(det.getTypeId().getName()))
SystemError: <built-in function cast> returned NULL without setting an exception

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2022-05-19:
This is the code of the unit test:
```
import Mesh
from pivy import coin

planarMesh = []
planarMesh.append( [-16.097176,-29.891157,15.987688] )
planarMesh.append( [-16.176304,-29.859991,15.947966] )
planarMesh.append( [-16.071451,-29.900553,15.912505] )
planarMesh.append( [-16.092241,-29.893408,16.020439] )
planarMesh.append( [-16.007210,-29.926180,15.967641] )
planarMesh.append( [-16.064457,-29.904951,16.090832] )

planarMeshObject = Mesh.Mesh(planarMesh)

Mesh.show(planarMeshObject)
view=FreeCADGui.ActiveDocument.ActiveView.getViewer()
rp=coin.SoRayPickAction(view.getSoRenderManager().getViewportRegion())
rp.setRay(coin.SbVec3f(-16.05,16.0,16.0),coin.SbVec3f(0,-1,0))
rp.apply(view.getSoRenderManager().getSceneGraph())
pp=rp.getPickedPoint()

det=pp.getDetail()
print (det)
print(det.getTypeId().getName())
det=coin.cast(det,str(det.getTypeId().getName()))

det.getFaceIndex() == 1
```

According to the error message it fails inside `det=coin.cast(det,str(det.getTypeId().getName()))` but I have no idea why. Could it be that your pivy is somehow broken? What's the output when you execute the Python code?

looooo, 2022-05-19:
If I run the code in the python console, I get the same issue:
```
>>> det=coin.cast(det,str(det.getTypeId().getName()))
Traceback (most recent call last):
  File "<input>", line 1, in <module>
SystemError: <built-in function cast> returned NULL without setting an error
```

luzpaz, 2022-05-25:
@looooo ping

looooo, 2022-05-27:
just tried again with python 3.9 and there i see this:
```
>>> det=coin.cast(det,str(det.getTypeId().getName()))Gui.runCommand('Std_OrthographicCamera',1)
>>> det.getFaceIndex() == 1
Traceback (most recent call last):
  File "<input>", line 1, in <module>
AttributeError: 'SoDetail' object has no attribute 'getFaceIndex'
>>> type(det)
<class 'pivy.coin.SoDetail'>
```

luzpaz, 2022-05-31:
@wwmayer :point_up:

wwmayer, 2022-05-31:
In C++ when you have picked a triangle then the pointer of SoDetail actually is an SoFaceDetail and one can do a type check and a static_cast or dynamic_cast.

With pivy the returned object always is an SoDetail and in order to get a SoFaceDetail the helper function coin.cast can be used. Therefore I still would like to know the result of `print(det.getTypeId().getName())` to be sure that the object is an SoFaceDetail. If yes, then there must be a problem with coin.cast.

luzpaz, 2022-06-05:
@looooo :point_up:

wwmayer, 2022-06-10:
Ok, then it looks like the function `coin.cast` has stopped working.

luzpaz, 2022-06-10:
@wwmayer shall we open a ticket in upstream coin3d repo ?

wwmayer, 2022-06-10:
Which pivy version does Conda use?

looooo, 2022-06-10:
I am using now 0.6.7 but I don't think there were any differences for coin.cast added to the pivy repo.

wwmayer, 2022-06-14:
What happens with:
```
from pivy import coin
det = coin.SoFaceDetail()
coin.cast(det, det.getTypeId().getName().getString())
```

With my locally build version from master where I reverted the latest changes of https://github.com/coin3d/pivy/commit/4b919a3f6b9f477d0e95a2fd2b6a149a55eb9792 I can confirm the error with:
```
from pivy import coin
det = coin.SoFaceDetail()
coin.cast(det, str(det.getTypeId().getName()))
```

See https://forum.freecadweb.org/viewtopic.php?p=602100#p602100

commit b85e9b1f (2022-06-14): Mesh: fixes #6902: Conda test errors
commit 6f0e0496 (2022-06-19): Mesh: fixes #6902: Conda test errors
closed 2022-06-14 by wwmayer