FreeCAD/FreeCAD #6908: [Feature Request] A workbench or tool for kinematic analysis of mechanisms (linkages)

FEA-eng, 2022-05-22:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=68826

### Subproject(s) affected?

Sketcher

### Idea description

It would be great to have a workbench or tool in FreeCAD that could make use of Sketcher's constraint capabilities to simulate 2D mechanisms (linkages) and calculate velocities, accelerations and possibly also forces based on the positions of the mechanism's points. Currently, this is only possible using scripting, as described in the referenced forum thread.

Such a tool was requested many times in the past. It would be very useful since there are pretty much no free/open-source solutions for such full kinematic analyses of 2D linkages at the moment.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-19:
Hmm, the issue tracker is also there fore feature requests but these should be doable by the community.

Wishing a completely new WB is way too much. Someone has to code it if he needs it. It is not sensible to just request this here. Nobody here could say "Nice, I have 2 free weeks and don't know what to do. Let's code a new WB!"

A sensible way is e.g. to contact developer you think that can do this, ask them and sponsor their work.

The feature requests here should be limited to e.g. missing features of existing WBs, missing core functionalities etc.
-> Closing, can be reopened when other developers agree it should stay open.

closed 2022-06-19 by donovaly

FEA-eng, 2022-06-19:
Then what about this FR ? #6439

Also, I didn't necessarily mean a new workbench. It could be as well a rather small tool to "drive" a sketch and read the positions of its points (which can already be done with a short Python script). Then just do some simple calculations to get the velocity, acceleration and so on.

Maybe this FR could be adjusted to make it clear that a single tool can be sufficient ?